Xinyuan-LilyGO/LilyGo-LoRa-Series #111: both lora and qmi spi conflict

Yasshion, 2023-11-07:
Hello, im trying to use both lora and qmi on t-beam s3 supreme board i had to disable the spi lora in the initboard function in order to make the QMI8658_GetDataExample work

void initBoard()
{
    Serial.begin(115200);
    Serial.println("initBoard");
    //SPI.begin(RADIO_SCLK_PIN, RADIO_MISO_PIN, RADIO_MOSI_PIN); if this is disable then qmi works

    Wire.begin(I2C_SDA, I2C_SCL);

#ifdef I2C1_SDA
    Wire1.begin(I2C1_SDA, I2C1_SCL);
#endif
in the end i want to use lora as well but when i activate lora

    if (!qmi.begin(IMU_CS)) {
        Serial.println("Failed to find QMI8658 - check your wiring!");
        while (1) {
            delay(1000);
        }
    } this line of code fails is there a possible solution for it because s3 has 4 and in the utilities.h radiospi and spi is suppose to be two different spi i think if it not ill try to define a new spi for it any other way can you help me to fix this issue?

lyusupov, 2023-11-14:
one can supply a spare SPI bus as an argument: for begin() method:
* https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/ESP32.cpp#L181
* https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/ESP32.cpp#L873-L878

lewisxhe, 2023-11-16:
Sorry, I have fixed it, please try againThe example passes the SPI object by default, but HSPI has been initialized as the SD card in the initBoard function, so for QMI, HSPI needs to be used for communication. This was my mistake. It has been fixed now.

lyusupov, 2023-11-23:
@Yasshion 

> Thank you, both of you it works now

If you are happy - would you mind to close this ticket, please ?

closed 2023-11-26 by lewisxhe

Bitbanger1900, 2024-01-15:
Hi,

I try to run the sample (QMI8658_WakeOnMotion) but it never works to use the QMI8658 over the Wire bus.
Over SPI it works but not over I2C.
Can you please provide me an example how to use the QMI with Wire?

Kind Regards,
Hannes

I uncomment the #define USE_WIRE and use the following to attach the QMI:

#define SPI_MOSI        35 
#define SPI_SCK         36
#define SPI_MISO        37
#define SPI_CS          47
#define IMU_CS          34
#define IMU_INT         33

#ifdef USE_WIRE
    //Using WIRE !!
    if (!qmi.begin(Wire, QMI8658_L_SLAVE_ADDRESS, SPI_MOSI, SPI_SCK)) {
        Serial.println("Failed to find QMI8658 - check your wiring!");
        while (1) {
            delay(1000);
        } 
    }

AXP2101 PMU init succeeded, using AXP2101 PMU
Power off and restart ALDO BLDO..
=========================================
DC1  : +   Voltage: 3300 mV 
DC2  : -   Voltage: 0900 mV 
DC3  : +   Voltage: 3300 mV 
DC4  : +   Voltage: 1840 mV 
DC5  : +   Voltage: 3300 mV 
ALDO1: +   Voltage: 3300 mV 
ALDO2: +   Voltage: 3300 mV 
ALDO3: +   Voltage: 3300 mV 
ALDO4: +   Voltage: 3300 mV 
BLDO1: +   Voltage: 3300 mV 
BLDO2: +   Voltage: 2800 mV 
=========================================
PowerKeyPressOffTime:4 Second
Started OLED
setupSDCard FAIL
Failed to find QMI8658 - check your wiring!

lewisxhe, 2024-01-16:
@Bitbanger1900 
On t-beam-s3, QMI is only connected to SPI, and I2C cannot be used.

Bitbanger1900, 2024-01-19:
Hi,

may i ask how Yasshion and you solved this with the SPI Bus Lora/QMI with an example? When QMI is working and i add 
an SPIClass with HSPI to start the LORA, QMI stops working as if there is always an SPI Bus conflict. Sometimes it works but after a powercycle most of the time not.