Xinyuan-LilyGO/LilyGo-LoRa-Series #113: Issue with LILYGO TBEAM AXP2101-V1.2 in Arduino IDE

vemanthchandrap, 2023-11-17:
Hello, this is my first project. I took two Lilygo tbeam axp2101-v1.2 for my Lora project. I need to find RSSI and SNR values. i have codes for transmitter and receiver and i am using LoRa library  by sandeep mistry in arduino ide. I used TTGO LoRa32 OLED as my board and i didn't get any error in the output, but i can't see any output values in serial monitor. anyone can help me with this.
output:
Sketch uses 280941 bytes (21%) of program storage space. Maximum is 1310720 bytes. 
Global variables use 22388 bytes (7%) of dynamic memory, leaving 272524 bytes for local variables. Maximum is 294912 bytes.
esptool.py v4.5.1
Serial port COM11
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:98:7a:6b:f7:a0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00054fff...
Compressed 17568 bytes to 12204...
Writing at 0x00001000... (100 %)
Wrote 17568 bytes (12204 compressed) at 0x00001000 in 0.5 seconds (effective 295.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 391.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 493.3 kbit/s)...
Hash of data verified.
Compressed 281312 bytes to 152630...
Writing at 0x00010000... (10 %)
Writing at 0x0001d278... (20 %)
Writing at 0x000250b1... (30 %)
Writing at 0x0002a328... (40 %)
Writing at 0x0002f84c... (50 %)
Writing at 0x00034dac... (60 %)
Writing at 0x0003d6a4... (70 %)
Writing at 0x0004812f... (80 %)
Writing at 0x0004d6ff... (90 %)
Writing at 0x00052ebb... (100 %)
Wrote 281312 bytes (152630 compressed) at 0x00010000 in 2.6 seconds (effective 881.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

lewisxhe, 2023-11-26:
Which sketch do you use?

vemanthchandrap, 2023-11-26:
Hey, the issue is solved. I used radiolib library. Thank you

closed 2023-11-26 by lewisxhe