Xinyuan-LilyGO/LilyGo-LoRa-Series #115: GPS backup power for warmstart

dodo124, 2023-11-20:
Hello and good morning,

I'm confronted with a new behaviour of my TTGO t-beam V1.1.

For the last couple of years it has worked perfectly as a TN mapper device.

But lately it cannot perform a warm start.

When I turn it off at 8pm and turn it on again at 6am, it has lost all GPS information.

The small battery next to the GPS module is almost empty and does not seem to be charged.

So should I do a factory reset or what else can I do?

Regards, Dom

dodo124, 2023-11-26:
> Can it be measured that the GPS module is powered?

Hi.
,

Yes it can be measured but it is very undty...

Greetings Dom

lewisxhe, 2023-11-26:
You can try to use this example, the results will be output in the Serial monitor. Please try
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/GPS/UBlox_Recovery/UBlox_Recovery.ino

dodo124, 2023-12-05:
html,body{background-color:#fff;color:#333;line-height:1.4;font-family:sans-serif,Arial,Verdana,Trebuchet MS;}Hi Lewis,
I followed the instructions but the problem still exists.
It is always cold starting, so it takes about one day or two to get gps.
No mater if there is a battery connected or not. Both batteries small and standby show 70-80% capacity but the last know location is lost.
:-(
Any ideas?
Last year it was working flawlessly.
Regards Dom
 
 
Lewis He schrieb am 26.11.2023 15:04 (GMT +01:00):

You can try to use this example, the results will be output in the Serial monitor. Please tryhttps://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/GPS/UBlox_Recovery/UBlox_Recovery.ino

closed 2024-03-28 by github-actions[bot]