Xinyuan-LilyGO/LilyGo-LoRa-Series #116: Lmic library not supporting SX1262 radio

rodri16, 2023-11-21:
Hi, have you tested SX1262 radio with Arduino LMIC? I am interested in TTGO LORA32S3 v1.2. Thanks

lewisxhe, 2023-11-26:
LMIC does not originally support SX1262. RadioLib already contains LoRaWAN examples, you can check it out.

rodri16, 2023-11-30:
Thanks @lewisxhe,
I cannot find examples you say.
LilyGo-LoRa-Series/lib/RadioLib/examples/SX126x no LoRaWAN examples

lewisxhe, 2023-11-30:
I am referring to the upstream RadioLib. After you are familiar with using local examples, you can switch to the mainline branch for development.

https://github.com/jgromes/RadioLib/tree/master/examples/LoRaWAN

rodri16, 2023-11-30:
So in Radiolib dev branch I could change radio to SX1262 and make example work? Will test it...
examples/LoRaWAN/LoRaWAN_End_Device_APB
SX1262 lora = new Module(NSS,DIO1,NRST,BUSY);

closed 2024-03-28 by github-actions[bot]