Xinyuan-LilyGO/LilyGo-LoRa-Series #117: T beam Supreme , upload sketch?

preshantram, 2023-11-25:
Hi, 

I am having a lot of difficulties to overwrite the default softRF with my own software.
Previously on my T-BEAM 1.2 it was rather easy via the IDE, but this simply don't want to work what so ever.
` PermissionError(13, 'A device attached to the system is not functioning.', None, 31)`.

So I tried a different approach by putting the device in the T-BEAM boot mode by pressing the BOOT short after RESET.
Now there is the 'TBEAMBOOT' device, which contain a CURRENT.UF2, I tried uploading my converted .BIN to UF2 by the  UF2conv.py, but it's always still comming up with the SOFTRF page.

What am I doing wrong, is there a way to upload via the arduino IDE, what about my UF2 approach?, how to update the sketch to my previous T-BEAM 1.2 sketch.

Thanks in advance.

lewisxhe, 2023-11-26:
Please follow the steps below to enter download mode correctly.
1. Connect the board via the USB cable
2. Press and hold the BOOT button , While still pressing the BOOT button, press RST
3. Release the RST
4. Release the BOOT button
5. Upload sketch

preshantram, 2023-11-26:
thanks for your reply, what about IDE settings for programming, can you please provide your settings?
Right now, I've figured out on how to upload my UF2, but now the media folder contains a FIRMWARE.BIN when I connect the unit to USB. When I upload a new BIN it changes the sketch, this is also fine.
But I don't want my firmware BIN to be so accessibel, is there any way to hide this?

> Please follow the steps below to enter download mode correctly.
> 
> 1. Connect the board via the USB cable
> 2. Press and hold the BOOT button , While still pressing the BOOT button, press RST
> 3. Release the RST
> 4. Release the BOOT button
> 5. Upload sketch

lewisxhe, 2023-11-28:
Instead of dragging the bin file directly into the disk, you need to convert the bin format to uf2 format before it can be used. For details, you can check [SoftRF](https://github.com/lyusupov/SoftRF/blob/7320acb740b38c69ebd9babd53f284c1f384e6d1/software/firmware/source/SoftRF/Makefile#L296)

closed 2024-03-28 by github-actions[bot]