Xinyuan-LilyGO/LilyGo-LoRa-Series #121: Lilygo ESP32 T3-S3 seems briked after upload

michelegirolami, 2023-12-05:
Dear LilyGo community,

- I ordered [lilygo ESP32 T3-S3](https://www.lilygo.cc/products/t3s3-v1-0) 
 the device comes with the meshtastic fw

- I managed to enter in the boot mode, by pressing the BOOT button while switching on the device
- I uploaded a custom fw (e.g. a blink fw) and the upload succeed as I can see the upload progress 
- after the upload completes, the device does not execute the new fw. The device can be put again in boot mode, but it does not start the new fw after RESETTING it . Below you can find the output from the serial port:
`ESP-ROM:esp32s3-20210327 Build:Mar 27 2021 rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT) Saved PC:0x4201d546 SPIWP:0xee mode:DIO, clock div:1 load:0x3fce3808,len:0x44c load:0x403c9700,len:0xbd8 load:0x403cc700,len:0x2a80 entry 0x403c98d0
`
I also noticed that if I switch off and on the device WITHOUT pressing the boot button, then I cannot see the assigned COM port (no COM port or /dev/tty port assigned to the device).

Any idea on how to restore the device so that to execute a custom fw? At the current stage, the device is bricked.

lewisxhe, 2023-12-05:
1.Connect the board via the USB cable
2.Press and hold the BOOT button , While still pressing the BOOT button, press RST
3.Release the RST
4.Release the BOOT button
5.Upload sketch
[image]

michelegirolami, 2023-12-05:
Dear   @lewisxhe thanks for your message. I appreciated it.
I executed your flow. I report below the upload process: 
```
Sketch uses 258013 bytes (19%) of program storage space. Maximum is 1310720 bytes.
Global variables use 18488 bytes (5%) of dynamic memory, leaving 309192 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port COM5
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: XXXXX
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0004ffff...
Compressed 15104 bytes to 10401...
Writing at 0x00000000... (100 %)
Wrote 15104 bytes (10401 compressed) at 0x00000000 in 0.3 seconds (effective 395.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 413.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 564.3 kbit/s)...
Hash of data verified.
Compressed 258384 bytes to 144168...
Writing at 0x00010000... (11 %)
Writing at 0x0001c738... (22 %)
Writing at 0x000245ce... (33 %)
Writing at 0x00029b46... (44 %)
Writing at 0x0002efbd... (55 %)
Writing at 0x00034654... (66 %)
Writing at 0x0003dc07... (77 %)
Writing at 0x00044e90... (88 %)
Writing at 0x0004a543... (100 %)
Wrote 258384 bytes (144168 compressed) at 0x00010000 in 2.3 seconds (effective 888.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

But the new fw is not loaded. When I press the RST button (after uploading the fw) the output from the serial is:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4202107a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
```

Any idea?

lewisxhe, 2023-12-05:
The board uses USB as the JTAG upload port. When printing serial port information on USB_CDC_ON_BOOT configuration needs to be turned on. If the port cannot be found when uploading the program or the USB has been used for other functions, the port does not appear. Please enter the upload mode manually.

michelegirolami, 2023-12-05:
Hi @lewisxhe you fixed my issue. Thanks a lot. Now I can upload and execute a custom fw.
But what is the USB_CDC_ON_BOOT  flag that I enabled? 
Thanks again!

lewisxhe, 2023-12-05:
Set up in Arduino, see my picture above

lyusupov, 2024-03-10:
> Hi @lewisxhe you fixed my issue.

[image]

closed 2024-03-10 by lewisxhe