Xinyuan-LilyGO/LilyGo-LoRa-Series #122: Connection between TTGO T-BEAM AXP2101 V1.2 and SD Card Module with spi interface

vemanthchandrap, 2023-12-12:
Good morning, 
I have tried to insert my TTGO T-BEAM AXP2101 V1.2 with a SPI SD card module. I will send you the code and pinouts I used to connect. I have run the code and there are no errors. But in the result SD is not detected. I used a 16GB SD card.
```

#include <RadioLib.h>
#include <SPI.h>
#include <SD.h>
#include "utilities.h"
#include "boards.h"

#define SD_CS_PIN 2 // Define the chip select pin for the SD card

SX1276 radio = new Module(RADIO_CS_PIN, RADIO_DIO0_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

File dataFile; // File object for handling SD card operations

// flag to indicate that a packet was received
volatile bool receivedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

void setFlag(void) {
    if (!enableInterrupt) {
        return;
    }
    receivedFlag = true;
}

void setup() {
    initBoard();
    delay(1500);

    // Initialize Serial communication
    Serial.begin(115200);

    // Initialize SD card
    if (!SD.begin(SD_CS_PIN)) {
        Serial.println(F("SD card initialization failed!"));
        while (true);
    }
    Serial.println(F("SD card initialized."));

    // initialize SX1276 with default settings
    Serial.print(F("[SX1276] Initializing ... "));
#ifndef LoRa_frequency
    int state = radio.begin(868.0);
#else
    int state = radio.begin(LoRa_frequency);
#endif
    if (state == RADIOLIB_ERR_NONE) {
        Serial.println(F("success!"));
        radio.setOutputPower(17);
        radio.setBandwidth(125);
        radio.setCurrentLimit(120);
    } else {
        Serial.print(F("failed, code "));
        Serial.println(state);
        while (true);
    }
    // set the function that will be called
    // when new packet is received
    radio.setDio0Action(setFlag, RISING);

    // start listening for LoRa packets
    Serial.print(F("[SX1276] Starting to listen ... "));
    state = radio.startReceive();
#ifdef HAS_DISPLAY
    if (u8g2) {
        if (state != RADIOLIB_ERR_NONE) {
            u8g2->clearBuffer();
            u8g2->drawStr(0, 12, "Initializing: FAIL!");
            u8g2->sendBuffer();
        }
    }
#endif
    if (state == RADIOLIB_ERR_NONE) {
        Serial.println(F("success!"));
    } else {
        Serial.print(F("failed, code "));
        Serial.println(state);
        while (true);
    }

    // if needed, 'listen' mode can be disabled by calling
      // any of the following methods:
      //
      // radio.standby()
      // radio.sleep()
      // radio.transmit();
      // radio.receive();
      // radio.readData();
      // radio.scanChannel();
}

void loop() {
    if (receivedFlag) {
        enableInterrupt = false;
        receivedFlag = false;

        String str;
        int state = radio.readData(str);

         if (state == RADIOLIB_ERR_NONE) {
            // packet was successfully received
            Serial.println(F("[SX1276] Received packet!"));

            // print data of the packet
            Serial.print(F("[SX1276] Data:\t\t"));
            Serial.println(str);

            // print RSSI (Received Signal Strength Indicator)
            Serial.print(F("[SX1276] RSSI:\t\t"));
            Serial.print(radio.getRSSI());
            Serial.println(F(" dBm"));

            // print SNR (Signal-to-Noise Ratio)
            Serial.print(F("[SX1276] SNR:\t\t"));
            Serial.print(radio.getSNR());
            Serial.println(F(" dB"));

            // print frequency error
            Serial.print(F("[SX1276] Frequency error:\t"));
            Serial.print(radio.getFrequencyError());
            Serial.println(F(" Hz"));
#ifdef HAS_DISPLAY
            if (u8g2) {
                u8g2->clearBuffer();
                char buf[256];
                u8g2->drawStr(0, 12, "Received OK!");
                u8g2->drawStr(5, 26, str.c_str());
                snprintf(buf, sizeof(buf), "RSSI:%.2f", radio.getRSSI());
                u8g2->drawStr(0, 40, buf);
                snprintf(buf, sizeof(buf), "SNR:%.2f", radio.getSNR());
                u8g2->drawStr(0, 54, buf);
                u8g2->sendBuffer();
            }
#endif
        } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
            // packet was received, but is malformed
            Serial.println(F("[SX1276] CRC error!"));

        } else {
            // some other error occurred
            Serial.print(F("[SX1276] Failed, code "));
            Serial.println(state);
        }

        if (state == RADIOLIB_ERR_NONE) {
            // Packet received successfully via LoRa

            // Write received data to the SD card
            dataFile = SD.open("data.txt", FILE_WRITE);
            if (dataFile) {
                dataFile.println(str);
                dataFile.close();
                Serial.println(F("[SD] Data written to SD card."));
            } else {
                Serial.println(F("[SD] Error opening file."));
            }

            // Rest of your processing and printing code
            // ...

            // Put module back to listen mode
            radio.startReceive();
            enableInterrupt = true;
        } else {
            // Error handling for LoRa packet reception
            // ...
        }
    }

    
}
```

PinOUTS: 5v-VCC, GND-GND,  GPIO 4 - MISO, GPIO 13 - MOSI, GPIO 14 - SCK, GPIO 2 - CS

Can you check the problem and suggest the solution. Thank you

lewisxhe, 2023-12-13:
The default SPI has been allocated for Radio use. For SD cards, you must specify the SPI Pin explicitly, for example
```
#define SDCARD_MOSI 15
#define SDCARD_MISO 2
#define SDCARD_SCLK 14
#define SDCARD_CS 13

SPIClass SDSPI(HSPI);
void setup()
{
     SDSPI.begin(SDCARD_SCLK, SDCARD_MISO, SDCARD_MOSI,);
if (!SD.begin(SDCARD_CS, SDSPI)) {
         Serial.println("setupSDCard FAIL");
     } else {
         uint32_t cardSize = SD.cardSize() / (1024 * 1024);
         Serial.print("setupSDCard PASS . SIZE = ");
         Serial.print(cardSize / 1024.0);
         Serial.println("GB");
     }
}
```

vemanthchandrap, 2023-12-13:
In the serial monitor, I am getting this 
Result:
PowPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drSD card initialization failed!
```
DC4  : -   Voltage: 1100 mV 
DC5  : -   Voltage: 1200 mV 
ALDO1: -   Voltage: 1800 mV 
ALDO2: +   Voltage: 3300 mV 
ALDO3: +   SD card initialization failed!
erKeyPressOffTime:4 Second
Started OLED
PowPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drSD card initialization failed!
SD card initialization failed!
```

I used 5V as VCC and the module is compatible.

lewisxhe, 2023-12-14:
1. Check the connection
2. Test with the simplest SD card example.
3. Take a photo to see if the connection is correct

closed 2024-03-27 by github-actions[bot]