Xinyuan-LilyGO/LilyGo-LoRa-Series #123: It does not install, ERROR 13: Library install failed: moving extracted archive to destination dir: Library not valid. Most recent version of IDE

BlueSkySunset76, 2023-12-16:


lewisxhe, 2024-01-03:
How do you move?

Zanctarios, 2024-01-07:
I am having the exact same problem! I tried using the .zip file downloading these files: https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master
And I click the Add Zip Library, under the **Sketch-> Include Library -> Add Zip Library option.**
Am I doing something wrong here?

Thanks in advance!

lewisxhe, 2024-01-08:
@Zanctarios This repository is not a library, just examples and resources
How to use please see the home page README

BlueSkySunset76, 2024-01-08:
That’s the correct procedure but it doesn’t work. I haven’t found the right way yet but I will and I’ll let you know as soon as I have it. Meanwhile they work well with other libraries and I’ve found them to be reliable.D:

closed 2024-03-26 by github-actions[bot]