Xinyuan-LilyGO/LilyGo-LoRa-Series #124: switch resistor for external WiFi Antenna T3_V1.6.1

maho24at, 2023-12-17:
Hi guys, 

which resistors do I have to change in case I want to use the ipx connector for WiFi? 

Board: T3_V1.6.1

thank you!!

[image]

[image]

lucaleicht, 2023-12-17:
I have the same question. But actually after a little research i think the ipex con we can find in yout second picture is just the second socket für the lora antenna (433/868/... MHz). If not, I'd be very interested too!

maho24at, 2023-12-18:
I also did some measurements, and looked at the tracing, I guess its just a false information on the website - maybe lost in translation... unfortunately...

lewisxhe, 2023-12-25:
[image]
Rotate 90° to the right

closed 2024-03-26 by github-actions[bot]