Xinyuan-LilyGO/LilyGo-LoRa-Series #126: LoRa V1.0 ESP32 SX1276 (868/915Mhz)

wolfgangmauer, 2024-01-01:
I got some of [these](https://de.aliexpress.com/item/32845370112.html) boards.
In the original flashed sketch, the blue LED flashes 1Hz.
But I can't find any(old) example source that does exactly that.

#define LILYGO_T3_V1_0

> LoRa Sender
> Sending packet: 0
> Sending packet: 1
> Sending packet: 2
> Sending packet: 3
> Sending packet: 4
> Sending packet: 5
> Sending packet: 6

Please guide me to the correct sample source!!!

THX

lewisxhe, 2024-01-02:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples%2FArduinoLoRa

wolfgangmauer, 2024-01-02:
THX!
The blue LED is at what Pin?
How to access the charging unit, if possible(detect charging)?

lewisxhe, 2024-01-03:
1. Charging instructions
2. Unable to detect

closed 2024-01-03 by wolfgangmauer