Xinyuan-LilyGO/LilyGo-LoRa-Series #127: LilyGo T3-S3: I2C ability?

Spy228, 2024-01-02:
Hello,

I have a question about I2C on this specific product T3-S3, V1.0, https://www.lilygo.cc/en-ca/products/t3s3-v1-0. I am a bit new to this so this may be a dumb question. 

I know the EPS32 has two buses of I2C available, and I believe the on-board display uses one of them, but I can't confirm. There isn't much documentation on this product.

I want to be able to add three I2C ADS1115 boards the the T3-S3 LilyGo dev-board, all on one I2C bus is fine with three different addresses which the expander boards can do, but I can't tell for sure that this dev-board has the ability or if the I2C buses are already used up?

Is someone able to confirm. Thank you.

lewisxhe, 2024-01-03:
Of course, OLED already occupies Wire
You can specify other GPIO as I2C, for example
```
Wire1.begin(15,16)
```

Spy228, 2024-01-05:
One more question, the sticker on the back of this board, T3-S3, is that just a mounting sticker, or is that for a heat-sink or something?

lewisxhe, 2024-01-06:
just a sticker

closed 2024-03-26 by github-actions[bot]