Xinyuan-LilyGO/LilyGo-LoRa-Series #128: Reducing deep sleep current for Lora T3S3 v1.2

redarflap, 2024-01-03:
Hi,

using the T3S3, i measure an average deep sleep current of about 2.5 mA, no matter if powered through USB or battery connector.

Are there any ways to further reduce power consumption? What would likely consume the most power on the board?
Would it help if I disconnect the Oled or other things, since I really only need Lora and one GPIO.

Alternatively, are there LilyGo-boards with LoRa that consume less power for future projects?

Best regards

lewisxhe, 2024-01-03:
1. Set the screen to sleep
2. Set LoRa to sleep
3. Test using battery power
4. Set ESP to ext1 external wake-up

redarflap, 2024-01-03:
@lewisxhe:

Thanks for your quick answer.

1./2.  Would I still need to set screen and LoRa to sleep if i never even initialized them? I'm testing with a basically empty ESP-IDF program right now.
3. I did. The consumption is very similar with battery and USB
4. I am not using any ext wakeup. I need the wakeup to be triggered by ULP

lewisxhe, 2024-01-04:
1.2 This is necessary. It is meaningless to set up sleep measurement consumption current without setting sleep measurement.

redarflap, 2024-01-05:
You were right. Initializing the OLED and SX1262 and then putting them to sleep brought the deep sleep power way down to somewhere between 20-100μA (hard to measure with my usb multimeter). Also powering the board through the battery connector reduced the current quite a bit.

I had to modify the SX126x and SSD1306 components i was using since none of them implemented the sleep feature, but it was straightforward.

Thanks for the tip.

closed 2024-01-05 by redarflap

balynt2048, 2024-03-07:
Hi, I have T3S3 v1.1 boards and I am also looking into the current consumption. I am using micropython and I initialized and sent to sleep both the OLED and the SX1262. I hooked up an ampmeter to a LiPo battery which I am using to power the board.
I can see that sending the OLED to sleep reduces amperage by 5 mA, and sending the Lora chip by 33 mA. I can also use full functionality of the components with my micropython drivers and I looked into the code and into the datasheets to make sure that I sent them to sleep as much as possible.

Still, after all componenets sleeping and sending the esp to deepsleep (with a timer wakeup), the lowest current I measure is 2.5 mA (lightsleep is about 6.5 mA).

Any tips what more I could do?