Xinyuan-LilyGO/LilyGo-LoRa-Series #129: Lilygo Lora32 T3_v1.6.1

ormingtrude, 2024-01-03:
I'm having problems with the Lilygo Lora32 T3_v1.6.1 Gouin pins using an ultrasonic sensor 

On the grip TXD, RXD 02,04,12, 13,14,15 pins I get "0cm" outputs on others (34,35,36,39 etc), i get - gpio_set_level(227): GPIO output gpio_num error. any advice would be appreciated

lewisxhe, 2024-01-04:
The maximum ESP32 GPIO is 39, and 227 is obviously wrong.

ormingtrude, 2024-01-04:
Thanks, any pointers on the zero output value, all I can think of is baudrate or faulty sensor?

I'm using 115200 as 9600 was just blank

lewisxhe, 2024-01-05:
Is Serial initialized correctly? Is it initialized like this?
```
Serial1.begin(9600, SERIAL_8N1,rxPin,txPin);
```
Can you post your test code and tell me how you connect?

I don't have the sensor, can I see if I can see the problem

ormingtrude, 2024-01-05:
Here’s the code I’m using to test the sensor:

#include <Wire.h>
#include <NewPing.h>

// Define pins for the ultrasonic sensor
#define triggerPin 21
#define echoPin 22
// Define the maximum distance for the sensor (in centimeters)
#define MAX_DISTANCE 800

// Define a NewPing object with the sensor's trigger and echo pins
NewPing sonar(triggerPin, echoPin, MAX_DISTANCE);

void setup() {
  Serial.begin(9600);
  Wire.begin(); // Initialize I2C communication

  // Initialize ultrasonic sensor pins
  pinMode(triggerPin, OUTPUT);
  pinMode(echoPin, INPUT);
}

void loop() {
  // Variables to hold the distance and duration
  unsigned int distance;
  unsigned long duration;

  // Send a ping to the sensor to get the distance
  duration = sonar.ping_median(5); // Median filter to improve accuracy

  // Debug print: Print the raw duration received from the sensor
  Serial.print("Raw duration: ");
  Serial.print(duration);
  Serial.println(" microseconds");

  // Calculate distance in centimeters
  distance = (duration / 2) / 29.1; // Calculate distance in centimeters

  // Debug print: Print the calculated distance
  Serial.print("Distance: ");
  Serial.print(distance);
  Serial.println(" cm");

  // Delay before next reading
  delay(5000);
}

I’m connecting to the pins per code with 5v and GND per board. The trig pin is resisted to drop voltage to 3.3v

Thx

lewisxhe, 2024-01-05:
21,22 have been used by OLED, and external pull-up resistors have been added by default. You cannot use these two IOs. You have to replace the IOs. You can try to use the following two Pins.

#include <Wire.h>
#include <NewPing.h>

// Define pins for the ultrasonic sensor
#define triggerPin 4
#define echoPin    36
// Define the maximum distance for the sensor (in centimeters)
#define MAX_DISTANCE 800

// Define a NewPing object with the sensor's trigger and echo pins
NewPing sonar(triggerPin, echoPin, MAX_DISTANCE);

void setup()
{
     Serial.begin(9600);

     // Default 21 = SDA, 22 = SCL
     Wire.begin(21,22); // Initialize I2C communication

     //Initialize ultrasonic sensor pins
     pinMode(triggerPin, OUTPUT);
     pinMode(echoPin, INPUT);
}

ormingtrude, 2024-01-05:
Thanks, I still get zeros.

I also tried serial.begin(115200)

New sensor is coming tomorrow, I will let you know

lewisxhe, 2024-01-05:
There are two types of AJ-SR04M sensors, one communicates through Serial and the other uses digital IO. Which one do you use?

ormingtrude, 2024-01-05:
Good question I think serial

Here's the product description 

Description
Specifications:
Working Voltage: DC5V
Quiescent Working Current: 5mA
Total Working Current: 30mA
Transmit Frequency: 40kHz
Max. Measuring Distance: 5m
Blind Zone: 25cm
Size: 28.5*41mm
Resolution: Approx 0.5cm
Angle: <50 degree
Working Temperature: -10℃~70℃
Storage Temperature: -20℃~80℃

Wiring:
+5V (power supply positive)
Trig (control) RX
Echo (receive) TX
GND (power supply negative)

Basic Working Principle:
(1) Using IO port TRIG trigger measuring distance, high level signal for at least 10uS.
(2) Module automatically sends eight 40kHz square , automatically detects whether there is return signal.
(3) When there is return signal, through IO port ECHO output a high level, the duration of the high level is the time of ultrasonic from sending to returning.Test distance = (high level time * sound velocity (340m/S))/2;

This module is simple to use, a port sends a high level of more than 10uS, then wait for high level output at receive port. Timer was enabled once there is output, and read time when this port becomes low level, which is the time of this measurement. Use the time to calculate distance. Constantly cycle measurement can reach your mobile measuring values.

The ultrasonic timing diagram shows that you only need to provide a more than 10uS pulse trigger signal, the module inside will send eight 40kHz cycle level and detect return . Output echo signal once detect return signal. The pulse width of echo signal is proportional to the measured distance.Thus calculate the distance with interval between sending signal and receiving echo signal.

Package Includes:
1 x Distance Measuring Module and Sensor

lewisxhe, 2024-01-06:
This is not Serial communication, this is communication through digital IO, so this has nothing to do with the baud rate.

ormingtrude, 2024-01-06:
Thanks Lewis I think, I need to do some research as I'm over my head a bit.

9600 yield nothing in print, whereas 115200 provides 0cm

I now have a spare new sensor but reluctant to try this as I'm concerned I might have damaged it with soldering or resisters.

Forgive me if the above is wrong or stupid

Thanks for patience and help so far. I'll check back in, in next few days.

ormingtrude, 2024-01-07:
You're awesome it works great. 

Thanks for your help, now onto next stage.

closed 2024-01-08 by ormingtrude