Xinyuan-LilyGO/LilyGo-LoRa-Series #131: T3S3v1.2 sx1280 RadioLibExample not working

53-74-65-66-61-6e, 2024-01-21:
Hello,

can someone confirm, that the Lilygo T3S3 v1.2 board works out of the box with the provided example ( [https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1280](url) ) ?

I am able to compile and upload the tx and rx example programs in vscode+platformio and the displays are working, but no data is received. I did some research in the RadioLib documentation, but no success so far.

To verify, that my boards are not broken, i can flash meshtastic and that defintiely works.

One thing, which does not seem to comply with RadioLib is the initalization of SPI. According to RadioLib, the SPI Class should be handed over in radio.begin() like here: [https://github.com/jgromes/RadioLib/wiki/Basics#non-standard-spi-setup](url)
I tried that as well, but still no luck.

Before i dive into more details, it would be helpful to know, if others are having success with this example on these boards.

best regards, Stefan

lewisxhe, 2024-01-22:
Please paste your test code

53-74-65-66-61-6e, 2024-01-22:
My test code is the example from Lilygo Github:
[examples/RadioLibExamples/SX1280](url)

53-74-65-66-61-6e, 2024-01-22:
i tried again step by step:
- download /LilyGo-LoRa-Series-master/
- open with platformio in vscode
- in platformio.ini uncomment: default_envs = T3_S3_V1_1_SX1280

1. in platformio.ini uncomment:
`src_dir = examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt`
- compilation and upload works. board boots and displays "Transmitting: OK!"

3. in platformio.ini uncomment:
`src_dir = examples/RadioLibExamples/SX1280/SX128x_Receive_Interrupt`
- compilation fails with errors:
`In file included from /LilyGo-LoRa-Series-master/examples/RadioLibExamples/SX1280/SX128x_Receive_Interrupt/SX128x_Receive_Interrupt.ino:28:
examples/RadioLibExamples/SX1280/SX128x_Receive_Interrupt/boards.h:116:15: error: redefinition of 'SPIClass SDSPI'
 SPIClass SDSPI(HSPI);

/LilyGo-LoRa-Series-master/examples/RadioLibExamples/SX1280/SX128x_Receive_Interrupt/SX128x_Receive_Interrupt.ino: In function 'void loop()':
/LilyGo-LoRa-Series-master/examples/RadioLibExamples/SX1280/SX128x_Receive_Interrupt/SX128x_Receive_Interrupt.ino:199:53: error: 'str' was not declared in this scope
                 snprintf(buf, sizeof(buf), "RX:%s", str);`

puboy7, 2024-01-23:
Please use the latest uploaded code. The specific problem is to accommodate the use of new screens and new versions of edp screens but to repeatedly define SDSPI at design time. Sorry for the inconvenience

53-74-65-66-61-6e, 2024-01-23:
Great, it works! [image]

Thank you very much for the direct support.

puboy7, 2024-01-23:
It's my pleasure.

53-74-65-66-61-6e, 2024-01-23:
one more thing. It would be nice, if the example shows the text which was send by the transmitter.

In [SX128x_Receive_Interrupt.ino](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/RadioLibExamples/SX1280/SX128x_Receive_Interrupt/SX128x_Receive_Interrupt.ino) use arduino string instead of counter:
```
// you can read received data as an Arduino String
String str;
int state = radio.readData(str);
/*
uint32_t counter;
int state = radio.readData((uint8_t *)&counter, 4);
*/
```
here at Serial print:
```
// print data of the packet
Serial.print(F("[SX1280] Data:\t\t"));
//Serial.println(counter);
Serial.println(str);
```
and here at oled output:
```
//snprintf(buf, sizeof(buf), "Data:%s", counter);
str = "Data: " + str;
str.toCharArray(buf, 256);
u8g2->drawStr(0, 26, buf);
```

53-74-65-66-61-6e, 2024-01-23:
[image]

puboy7, 2024-01-23:
Yes, as you can see in the sample code for other models there are also examples that show you what you want according to your requirements

puboy7, 2024-01-24:
Our original plan was to remove the use of pa mode in version 1.2 but later added it back so please try it with the latest code

53-74-65-66-61-6e, 2024-01-24:
I was confused, because in the schematics, there is no lora_tx and lora_rx at U1 HDP17.

Are these pins unique to the PA versions of the chips?

puboy7, 2024-01-25:
The PA version does not have lora_tx and lora_rx so you need to define it by yourself

closed 2024-03-26 by github-actions[bot]