Xinyuan-LilyGO/LilyGo-LoRa-Series #132: /schematic/T3_S3_V1.2.pdf conflicting pin numbers?

53-74-65-66-61-6e, 2024-01-22:
the schematics for T3_S3_V1.2 ( [/schematic/T3_S3_V1.2.pdf](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/T3_S3_V1.2.pdf)
 ) have to different pin numbers for sx1280 DIO1 and sx1280 BUSY:
- according to the drawing it should be:
sx1280 (U1, HPD17) DIO1 : IO9
sx1280 (U1, HPD17) BUSY : IO36

- according to the textbox on the right side of the drawing ( [see detail from page 2](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/assets/157323054/2c71edec-e3e3-4934-94c3-6b2b0d4e7301) ):
sx1280 DIO1 : IO33
sx1280 BUSY : IO34

- which version is right?

puboy7, 2024-01-23:
All pin definitions are based on the schematic diagram, the annotations beside them are wrong,
and the first set is the correct pin number ：
(sx1280 (U1, HPD17) DIO1: IO9
sx1280 (U1, HPD17) BUSY: IO36)
Sorry for the inconvenience

53-74-65-66-61-6e, 2024-01-23:
thanks for the reply!

closed 2024-03-26 by github-actions[bot]