Xinyuan-LilyGO/LilyGo-LoRa-Series #133: Lilygo T3-S3: SPI Pins?

VTEca, 2024-01-29:
Hello,

I have a question about SPI bus on this specific product T3-S3, V1.2, https://www.lilygo.cc/en-ca/products/t3s3-v1-0. I am a bit new to this so this may be a dumb question.

I would like to connect this chip: 16ch SPI ADS7953 which communicates via SPI, but I don't know which pins to connect it to. 

On the T3-S3 schematic it looks like the SPI pins are not connected? Is it possible to connect this chip to the T3S3? If so, can you please recommend which pins to use?

[image]

Thank you!

lewisxhe, 2024-01-30:
The SPI Pin can be freely mapped. The one you circle is the internal flash pin and cannot be used externally.

If you want to use SPI, then you can refer to how to initialize the SD card,
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/45b66f48e1fff1821f01638f474ef377dc520d34/examples/RadioLibExamples/SX1276/SX1276_Receive_Interrupt/boards.h#L411
You only need to pass in MISO, MOSI, SCK during initialization to complete the initialization of SPI. Please note that the default SPI has been assigned to Radio. If you need to assign additional SPI to external devices, you need to use other SPI For example the following example

SPIClass mySPI(HSPI);
mySPI.begin(sck, miso, mosi,);

sck,miso,mosi , You need to decide based on Pinmap or schematic, it must be free, or you can share SPI bus with SD card

Tom-E, 2024-03-07:
On T3S3, as I understand, HSPI=1 is already mapped to radio https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/45b66f48e1fff1821f01638f474ef377dc520d34/examples/RadioLibExamples/SX1262/SX1262_Receive_Interrupt/boards.h#L10 which means that FSPI=0 is the only other option. I am not using SD card, is there anything besides:

```
SPIClass mySPI(0); // HSPI=1 is used by radio, FSPI=0 which we dont use flash
mySPI.begin(CLK_PIN, DOUT_PIN, DIN_PIN, CS_PIN);
```

Are there any issues with using these GPIO pins?
```// SPI Settings for MCP3204
const int CLK_PIN = 38;  // SPI Clock
const int DOUT_PIN = 43; // MISO
const int DIN_PIN = 44;  // MOSI
const int CS_PIN = 39; // Chip Select for MCP3204 #1
```

Right now I'm getting data back when I read, but the data is nonsensical, so not sure if it's my read code or something with the re-use of SD card SPI Bus...

closed 2024-04-20 by github-actions[bot]