Xinyuan-LilyGO/LilyGo-LoRa-Series #136: LilyGo T3S3 SX1262 Serial Monitor not working

Zanctarios, 2024-02-19:
Hello everyone,
I recently bought two LilyGo T3S3 SX1262 boards and I've been using the examples provided on this Github. I have been successful on programming a receiving and transmitting side, however the transmitting board never worked on Arduino IDE's Serial Monitor. It was rather odd, but I ignored it since I didn't need the Serial Monitor on the transmitter at that point in time.
The only thing I kept seeing the Serial Monitor was this:
[image]

Afterwards, I needed to modify the Spreading Factor to do some tests. To debug the code I needed the Serial Monitor. It never worked on the transmitter, so I programmed it on the receiver board and it failed on it too! After that, even after programming the receiver back to receiving mode I haven't been able to get anything on either board from the Serial Monitor. It is driving me crazy and I don't know what else to do! I had one board that could connect to the Serial Monitor and now I cannot connect on either of them.

Here are my settings on the programmer on Arduino if someone needs them:
[image]

I'm doing some research for my master's degree and I really need this to work, can someone help me out with this?

Thanks in advance for any advice,
Zanctarios

lewisxhe, 2024-02-20:
```
Tools -> USB CDC On Boot -> Enable
```

Zanctarios, 2024-02-20:
Thanks for the answer, this solved it!

closed 2024-02-20 by Zanctarios