Xinyuan-LilyGO/LilyGo-LoRa-Series #137: T beam supreme bricked?

preshantram, 2024-02-19:
Hi, I've uploaded a firmware with the JTAG debug on,
Now I cannot upload a firmware anymore, previously I could press the boot button shortly after reset which allowed me to upload a new firmware, but I cannot put my T beam Supreme in this option anymore, boot button doesn't seem to do a thing.

Any ideas??

lewisxhe, 2024-02-20:
[image]

1. Connect the board via the USB cable
2. Press and hold the BOOT button , While still pressing the BOOT button, press RST
3. Release the RST
4. Release the BOOT button
5. Upload sketch

lyusupov, 2024-02-24:
> previously I could press the boot button shortly after reset which allowed me to upload a new firmware, but I cannot put my T beam Supreme in this option anymore

https://github.com/adafruit/tinyuf2/tree/master/ports/espressif

closed 2024-04-09 by github-actions[bot]