Xinyuan-LilyGO/LilyGo-LoRa-Series #140: How tu use SDCARD on T-Beam S3 Supreme ?

ric-tim, 2024-03-06:
I'm looking to use the SD Card in the T-Beam Supreme.
According to the documentation, everything should work with these values:
#define SPI_MOSI (35)
#define SPI_SCK (36)
#define SPI_MISO (37)
#define SPI_CS (47)

SPIClass hSPI(HSPI);
....
void setup(){
    Serial.begin(115200);
    hSPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI,SPI_CS);
    if(!SD.begin(SPI_CS,hSPI)){
        Serial.println("Card Mount Failed");
        return;
    }
....
Still "Card Mount Failed" even though the QMI8658 on the same SPI bus works fine with the https://github.com/lewisxhe/SensorsLib library and IMU_CS (34).

What else do I need to do?
Thanks for your help....

lewisxhe, 2024-03-07:
This example cannot mount SD?

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/Sensor/QMI8658_GetDataExample/QMI8658_GetDataExample.ino

ric-tim, 2024-03-07:
Thank you for your reply.
qmi.begin() does give access to the SPI bus and allows you to use QMI8658, but how can you take advantage of this to mount the SD and use the filesystem? 
sd.begin(SPI_CS) from the sdfat.h library tries to mount the SD at the same time as the filesystem and fails.

lewisxhe, 2024-03-08:
Are you using https://github.com/adafruit/SdFat? Isn’t it the <SD.h> that comes with esp32?

ric-tim, 2024-03-09:
OK. Solved ! Thanks for everything.
The important hint in Lewis He's remarks proposing QMI8658_GetDataExamples is doing initBoards() which could be limited to initPMU().  Indispensable, if I've understood correctly, for using TF Card on T-Beam S3 Core!

lyusupov, 2024-03-10:
[image]

closed 2024-03-10 by ric-tim