Xinyuan-LilyGO/LilyGo-LoRa-Series #141: Does the T3-S3 have SPIRAM?

antirez, 2024-03-07:
Hello. `esptool` reports 2MB SPIRAM that would be fantastic to use, however the pins that are usually needed to access such SPIRAM here are used for the LoRa chip, so: does the ESP32-S3 in T3-S3 lacks the SPRIAM or the SPIRAM is connected to other pins? Thanks.

balynt2048, 2024-03-07:
It definitely has SPIRAM (2MB) on my board which is a T3S3 v1.1 with the SX1282 lora modem. I cannot say which ESP pins were used for this.

antirez, 2024-03-07:
> It definitely has SPIRAM (2MB) on my board which is a T3S3 v1.1 with the SX1282 lora modem. I cannot say which ESP pins were used for this.

Interesting! I got a v1.2, and MicroPython is not able to recognize the SPIRAM. I guess it's just a matter of selecting the right pins then. Let's see if LILYGO has some information. What software are you using to see the SPIRAM?

balynt2048, 2024-03-07:
I'm also using micropython. You have to recompile the micropython firmware to use it. See https://github.com/orgs/micropython/discussions/10156

lewisxhe, 2024-03-07:
@antirez T3-S3 uses QSPI type PSRAM. In Micropython, you must burn the PSRAM type to QSPI interface before it can run. ESP32S3 has two PSRAM interfaces, one is OPI (eight lines) and the other is regular 4-line SPI PSRAM. (QSPI)

balynt2048, 2024-03-07:
I'm using it with micropython without a problem. You have to get the proper firmware for it.
What I had to do is recompile with some custom settings. See https://github.com/orgs/micropython/discussions/10156
Specifically, I had to set the flash size to 4MB and made sure that I am building a "GENERIC_S3_SPIRAM".

antirez, 2024-03-07:
> @antirez T3-S3 uses QSPI type PSRAM. In Micropython, you must burn the PSRAM type to QSPI interface before it can run. ESP32S3 has two PSRAM interfaces, one is OPI (eight lines) and the other is regular 4-line SPI PSRAM. (QSPI)

Thank you @lewisxhe, indeed I'm using the 8 line build right now. I'll switch to QSPI. Thanks.

antirez, 2024-03-07:
> I'm using it with micropython without a problem. You have to get the proper firmware for it. What I had to do is recompile with some custom settings. See [...]

Yep I know but if it was just for the 4MB flash, there is a command line utility that changes the image in a second without any need to recompile. So the image is working great, if not for SPIRAM, as I was using the 8 line version.

antirez, 2024-03-07:
Ok now I can see all the memory available. Huge amount! Thanks.

closed 2024-03-08 by antirez