Xinyuan-LilyGO/LilyGo-LoRa-Series #142: [Issue] Example LoRaSender do not work on T-Beam V1_2

redradist, 2024-03-08:
I have tried to run the following example https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/ArduinoLoRa/LoRaSender/LoRaSender.ino, but on `LoRa.endPacket();` program stuck ...

I have deep dive that figured out that it spinning in while loop inside `endPacket` method:
```c++
int LoRaClass::endPacket(bool async)
{
...
  Serial.println("endPacket after: if (!async)");
  if (!async) {
    Serial.println("endPacket before: while ((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0)");
    // wait for TX done
    while ((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0) {
      int reg = readRegister(REG_IRQ_FLAGS);
      yield();
    }
    Serial.println("endPacket after: while ((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0)");
    // clear IRQ's
    writeRegister(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);
  }
  Serial.println("endPacket finish");
...
}
```

This is the same issue that I mentioned in https://github.com/LilyGO/TTGO-T-Beam/issues/60

lewisxhe, 2024-03-11:
[ArduinoLoRa](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/ArduinoLoRa) 
Only supports SX1276/SX1278, which one are you using?

lewisxhe, 2024-03-11:
Although SX1276 is designed for 868MHZ, it should not be stuck here when using 433MHZ. Have you changed the frequency and tested it? For example, 868MHZ, or test the RadioLib example?

redradist, 2024-03-11:
@lewisxhe Yes, I have verified it with all frequencies: 433E6,470E6,868E6,915E6
Here is my code that I used to run app:
```cpp
#include <LoRa.h>
#include "boards.h"

int counter = 0;

void setup()
{
    initBoard();
    // When the power is turned on, a delay is required.
    delay(1500);

    SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);

    Serial.println("LoRa Sender");
    LoRa.setPins(LORA_CS, LORA_RST, RADIO_DIO0_PIN);
    if (!LoRa.begin(LoRa_frequency)) {
        Serial.println("Starting LoRa failed!");
        while (1);
    }
    LoRa.setTxPower(17);
    Serial.println("LoRa started successfully");
}

void loop()
{
    Serial.print("Sending packet: ");
    Serial.println(counter);

    // send packet
    int status = LoRa.beginPacket();
    Serial.print("Is beginPacket started ");
    Serial.println(status);
    Serial.println("After LoRa.beginPacket()");
    for (int i = 0; i < 100; ++i) {
        LoRa.print("hello ");
        LoRa.print(counter);
    }
    Serial.println("After counter");
    int endStatus = LoRa.endPacket();
    Serial.print("Is endPacket started ");
    Serial.println(endStatus);

    Serial.print("Finish sending packet: ");

    counter++;
    delay(5000);
}
```

lewisxhe, 2024-03-13:
If there is no problem with the sketch, then the possible problem is that there is a problem with the LoRa module. Contact the seller for replacement.

lewisxhe, 2024-03-13:
Finally, you may need to conduct a test, write a test file, and then after starting, press the button IO38 to switch the function page to see if the value of the TX page will increase. If there is still no change, then there is undoubtedly a hardware problem.
[firmware](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/hardware_test/t-beam-v1.x/SX1276)

redradist, 2024-03-13:
@lewisxhe 
> If there is no problem with the sketch, then the possible problem is that there is a problem with the LoRa module. Contact the seller for replacement.

I think it is not a chip problem, because Mechanistic firmware works properly

redradist, 2024-03-13:
@lewisxhe 
> Finally, you may need to conduct a test, write a test file, and then after starting, press the button IO38 to switch the function page to see if the value of the TX page will increase. If there is [...]

I have the following 433Mhz T-Beam:
[image]

I have wrote the following code using RadioLib:
```cpp
#include <RadioLib.h>
#include <SPI.h>
#include <TinyGPS++.h>
#include <memory>

#define LORA_SS         18
#define LORA_DIO0       26
#define LORA_DIO1       33
#define LORA_DIO2       32
#define LORA_RST        23

#define FREQUENCY       433

TinyGPSPlus GPS;

#define BUTTON_PIN 38

template<typename TRadioChip>
void initRadioLib(std::string name, uint32_t freq, uint32_t pinMode) {
    auto radio = std::unique_ptr<TRadioChip>(new TRadioChip(new Module(LORA_CS, LORA_IRQ, LORA_RST, pinMode)));
    int state = radio->begin();
    Serial.print(F("initRadioLib: "));
    Serial.print(F("name = "));
    Serial.print(name.c_str());
    Serial.print(F(", freq = "));
    Serial.print(freq);
    Serial.print(F(", pinMode = "));
    Serial.print(pinMode);
    if (state == RADIOLIB_ERR_NONE) {
        Serial.println(F(". LoRa success !!"));
        radio->setOutputPower(17);
    } else {
        Serial.print(F(". LoRa failed, code "));
        Serial.println(state);
    }
}

void verifyParams(uint32_t freq, uint32_t pinMode) {
    SPI.setFrequency(freq);
    initRadioLib<SX1261>("SX1261", freq, pinMode); // 1
    initRadioLib<SX1262>("SX1262", freq, pinMode); // 2
    initRadioLib<SX1268>("SX1268", freq, pinMode); // 3
    initRadioLib<SX1272>("SX1272", freq, pinMode); // 4
    initRadioLib<SX1273>("SX1273", freq, pinMode); // 5
    initRadioLib<SX1272>("SX1272", freq, pinMode); // 4
    initRadioLib<SX1276>("SX1276", freq, pinMode); // 6
    initRadioLib<SX1272>("SX1272", freq, pinMode); // 4
    initRadioLib<SX1277>("SX1277", freq, pinMode); // 7
    initRadioLib<SX1278>("SX1278", freq, pinMode); // 8
    initRadioLib<SX1279>("SX1279", freq, pinMode); // 9
    initRadioLib<SX1280>("SX1280", freq, pinMode); // 10
    initRadioLib<SX1281>("SX1281", freq, pinMode); // 11
    initRadioLib<SX1282>("SX1282", freq, pinMode); // 12
}

void setup() {
    Serial.begin(115200);
    Serial1.begin(9600, SERIAL_8N1, 34, 12);   //17-TX 18-RX for GPS

    // Very important for SPI pin & LoRa configuration!
    pinMode(BUILTIN_LED, OUTPUT); // For LED feedback
    pinMode(BUTTON_PIN, INPUT); // Middle button next to LoRa chip. The one on the right is RESET, careful...
[... 67 lines cut ...]
initRadioLib: name = SX1280, freq = 9600, pinMode = 32. LoRa failed, code -705
initRadioLib: name = SX1281, freq = 9600, pinMode = 32. LoRa failed, code -705
initRadioLib: name = SX1282, freq = 9600, pinMode = 32. LoRa failed, code -705
initRadioLib: name = SX1261, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1262, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1268, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1272, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1273, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1272, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1276, freq = 115200, pinMode = 26. LoRa success !!
initRadioLib: name = SX1272, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1277, freq = 115200, pinMode = 26. LoRa success !!
initRadioLib: name = SX1278, freq = 115200, pinMode = 26. LoRa success !!
initRadioLib: name = SX1279, freq = 115200, pinMode = 26. LoRa success !!
initRadioLib: name = SX1280, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1281, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1282, freq = 115200, pinMode = 26. LoRa failed, code -2
initRadioLib: name = SX1261, freq = 115200, pinMode = 33. LoRa failed, code -2
```

It is strange ...
Looks like I need some how to reset LORA32 module at the start of board ...
Is there any mechanism for it ??

Also according what I have found, why then LORA32 433MHz module do not work with standard example ?

lewisxhe, 2024-03-14:
The one with 433M on the picture is definitely SX1278, not SX1276

redradist, 2024-03-14:
> The one with 433M on the picture is definitely SX1278, not SX1276. Regardless of whether it is powered by battery or not, the power needs to be turned on before using LoRa, which is controlled by PMU.
> 
> https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/43e93a7e654803b48962702e07fdf275b9d6b6fe/examples/RadioLibExamples/SX1278/SX1278_Transmit_Interrupt/boards.h#L171

@lewisxhe Thanks a lot !!
The following example works properly https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1278/SX1278_Transmit

... but I still have question, why then https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/ArduinoLoRa/LoRaSender did not works properly ?

Also how do you know that this board has SX1278 and not SX1276 ?? Both chips could work on 433M ...
Is there any documentation ?
On official site https://www.lilygo.cc/products/t-beam-softrf?variant=43170136948917 mentioned that it could be in configuration of 3 different chips: SX1262, SX1276, SX1278 ...

lewisxhe, 2024-03-14:
> but I still have question, why then https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/ArduinoLoRa/LoRaSender did not works properly ?
I've tested this and found no problems, and I don't know what the problem is.

>Also how do you know that this board has SX1278 and not SX1276 ?? Both chips could work on 433M ...
Is there any documentation ?
On official site https://www.lilygo.cc/products/t-beam-softrf?variant=43170136948917 mentioned that it could be in configuration of 3 different chips: SX1262, SX1276, SX1278 ...
README FAQ mentioned in point 7. If the label says 433MHZ, then it is SX1278

redradist, 2024-03-14:
> I've tested this and found no problems, and I don't know what the problem is.
>
>
> README FAQ mentioned in point 7. If the label says 433MHZ, then it is SX1278

@lewisxhe I see that according this screenshot it is possible to buy 433MHZ with SX1262 chip:
[image]

lewisxhe, 2024-03-14:
Yes, check your order to see which one you bought

redradist, 2024-03-14:
> Yes, check your order to see which one you bought

@lewisxhe Unfortunately I've bought from reseller and this information is not mentioned there ...
Is there a way to detect programmatically which chip I have ? Maybe I could read some register value for it ?

lewisxhe, 2024-03-14:
You can flash [meshtastic](https://meshtastic.org/downloads/) to view the startup log to see which chip the firmware detects.

redradist, 2024-03-21:
Okay, thanks, I think issue could be closed

closed 2024-03-21 by redradist