Xinyuan-LilyGO/LilyGo-LoRa-Series #145: Unable to upload sketch

Wellco, 2024-03-21:
I have the T-Beam supreme S,  and before I start coding I wanted to test everything is as expected. So I decided to upload the KitchenSink example sketch. The sketch compiles ok, but never uploads. I get this error:

Failed uploading: uploading error: exit status 2

Now I think the device is not in the right mode to receive the new upload, but how do you get it into the mode?  Does the device have to be in T-Beam Boot modeI have read several post here and elsewhere saying you need to press and hold the Boot and RST button while before/after plugging in the USB cable.

If this is the solution to my issue, can someone confirm what is the correct sequence of button pushes, as I have only reach that mode twice. And fail to do it every time.

When I have been in T-Beam Boot mode, I see device is mounted as a drive on my Mac. When in this mode usb connection used by IDE is not available, and I guess that's one of the reasons I get the error..

I have also read that you need to convert your compiled sketch bin file into a UF2 file. Never done this before and reading on this so far has me scratching my head. Is there a simple way to do this?

Any advice is much appreciated.

lewisxhe, 2024-03-24:
README -> FAQ -> [Point 4](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series?tab=readme-ov-file#7%EF%B8%8F%E2%83%A3-faq)

Wellco, 2024-03-24:
Interesting since I am sure I did those steps, quite a few times in fact. It seems that after I got rid of SoftRF by using VS code instead of Arduino IDE, I am now able to connect with either tool.  

Thanks for your help

closed 2024-03-24 by Wellco