Xinyuan-LilyGO/LilyGo-LoRa-Series #147: Powering options

nicklasb, 2024-03-23:
Hi,

I cannot find somewhere authoritative where the different power options for the T-Beam v1.1 is described. 

That it can be powered using USB is obvious, but it is not clear if the battery is charged if power is provided through the 5v pin, for example, "someone on the internet" think that it won't be charged then, which would narrow the possible usages of the board.
Power using USB is certainly not always an option, at least not in my use case.

lewisxhe, 2024-03-24:
Cannot be charged via 5V Pin.

nicklasb, 2024-03-24:
> Cannot be charged via 5V Pin.

Does that mean that the only way to charge it is using the USB? 
Is there no other way but adding an external charge controller?

lewisxhe, 2024-03-26:
When T-Beam v1.1 was first designed, there was no external power supply input interface reserved other than USB. If you don’t want to connect the power supply from USB, you can connect the power supply from the arrow in the picture below. This is the same as the USB input. A line, plugged in here, means USB power is enabled.

[image]

StijnVerhelst, 2024-03-26:
> When T-Beam v1.1 was first designed, there was no external power supply input interface reserved other than USB. If you don’t want to connect the power supply from USB, you can connect the power [...]

I have a similar use-case, are we allowed to connect the 5V pin to the capacitor?

lewisxhe, 2024-03-27:
Of course, please remember that the location of this capacitor is the location of the USB 5V input.

nicklasb, 2024-03-27:
> When T-Beam v1.1 was first designed, there was no external power supply input interface reserved other than USB.

Does this indicate that there are less hacky methods for the 1.2?

closed 2024-05-12 by github-actions[bot]