Xinyuan-LilyGO/LilyGo-LoRa-Series #148: paxcounter dont work on LoRa32 V2.1_1.6 TCXO

nachoplus, 2024-03-24:
LMIC use DIO1 but in this version is not connected. See https://github.com/cyberman54/ESP32-Paxcounter/issues/1005

¿there is any workarround?

lewisxhe, 2024-03-24:
I need to confirm again

lewisxhe, 2024-03-25:
It has been confirmed that DIO1 is not led out, IO33 is the Radio TXCO crystal enable, and IO33 must be set to HIGH before use.

nachoplus, 2024-03-25:
But  DIO1 is mandatory for LMIC library so this board is **unusable for any  lorawan project**.  This include Pax counter, ttn_mapper as well  any other TTN projects. Great limitation. In my opinion this must be warned clearly in the product pages. 

I recently buy two of this modules for LoraWan projects. This is why I ask for a workarround.

lewisxhe, 2024-03-25:
OK, I will give feedback.

nachoplus, 2024-04-05:
Hello, any solution for this issue?

lewisxhe, 2024-04-06:
The current version cannot solve this problem. DIO1 will be reconnected to ESP in subsequent versions.

cyberman54, 2024-04-14:
@lewisxhe While DIO1 is not connected, the TCXO version shouldn't be offered in the LilyGo aliexpress shop for Paxcounter (and other LoRaWAN projects). Same for 433 MHz version, since there is no 433 MHz support in LMIC yet.

lewisxhe, 2024-04-15:
@cyberman54 Yes, I have reported this issue to LilyGo, and they should have removed it from the shelves now. The version with DIO1 will be withdrawn later.

cyberman54, 2024-05-19:
This issue can be closed.

closed 2024-07-04 by github-actions[bot]