Xinyuan-LilyGO/LilyGo-LoRa-Series #149: How to put LoRa32 V2.1_1.6 in boot mode for flashing

bennett-elder, 2024-03-24:
Hello,

I tried finding instructions for putting LoRa32 V2.1_1.6 in boot mode so I can flash it with new firmware. I bought it recently with disaster radio on it but want to flash it to meshtastic. It's not showing up when connected via a USB data cable.

Found some threads that seemed to point to shorting GND to IO0. They said make sure only the red light comes on, not the flashing blue one. When I short GND to IO0 and then plug in via USB they both come on. Am I shorting the correct pin to ground? Is there another process I should follow?

lewisxhe, 2024-03-24:
README -> FAQ -> [Point 4](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series?tab=readme-ov-file#7%EF%B8%8F%E2%83%A3-faq)

Wellco, 2024-03-24:
@bennett-elder I had similar issues and read about shorting the GND to IO0, but I think that's only if your board doesn't have a reset button.

Also I found that using PlatromIO in VS code I was able to upload my sketch, where I couldn't with IDE. Once I uploaded in VS code I was then able to upload from IDE. Seems a bit strange, but yes you do need to follow point 4 above.

dmfrey, 2024-03-26:
I'm trying to reset the board and nothing ever happens. I've connected one of the GND ports with IO0, plug it in, press and release RST, but nothing. Am I doing something wrong? The 2 lights on the board are showing Red(left) and Blue(Right). Do they mean anything?

dmfrey, 2024-03-26:
Disregard my last message. It did allow me to upload a new firmware. The screen is blank, so I was thinking it was broke.

lewisxhe, 2024-03-27:
Which sketch did you use for testing?
Is there any visible damage to the screen?

dmfrey, 2024-03-27:
It was my mistake. I was getting no output on the screen, but that was by design on the program I was using. I was able to flash it correctly and it's up and running on my network and connected to my Home Assistant installation.

dmfrey, 2024-03-27:
closes #149

bennett-elder, 2024-04-01:
> README -> FAQ -> [Point 4](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series?tab=readme-ov-file#7%EF%B8%8F%E2%83%A3-faq)

Following up to note that that did not put it in boot mode when I tried it that day.

If anyone has put specifically LoRa32 V2.1_1.6 (no, there’s no boot button) in boot mode, I’d like to hear how. In the meantime, I’ve moved onto different hardware.

closed 2024-05-16 by github-actions[bot]