Xinyuan-LilyGO/LilyGo-LoRa-Series #150: T3S3 v1.0 DIO1 issue

TomsGitBits, 2024-03-25:
I am attempting to run a [basic Lora transmitter example](https://github.com/StuartsProjects/SX12XX-LoRa/blob/master/examples/SX128x_examples/Basics/3_LoRa_Transmitter/3_LoRa_Transmitter.ino) from the [StuartsProjects/SX12XX-LoRa library](https://github.com/StuartsProjects/SX12XX-LoRa/tree/master). I know the LilyGo library has a basic transmitter example of it's own, which I've got working, but the StuartsProjects library has some other specific examples applicable to my use case that I'm hoping to get working eventually. This example is just my starting point.

I mapped the pin definitions over from the [SX128x_Transmit_Interrupt](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt) example at [defined(LILYGO_T3_S3_V1_0)](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/83dd32ad95fe6a19c67065f4d33bbb72004d3360/examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt/utilities.h#L178). This resulted in the below settings for the StuartsProjects example:
```
/*******************************************************************************************************
  Programs for Arduino - Copyright of the author Stuart Robinson - 06/02/20

  This program is supplied as is, it is up to the user of the program to decide if the program is
  suitable for the intended purpose and free from errors.
*******************************************************************************************************/

//*******  Setup hardware pin definitions here ! ***************

//These are the pin definitions for one of my own boards, the Easy Pro Mini,
//be sure to change the definitions to match your own setup. Some pins such as DIO2,
//DIO3, BUZZER may not be in used by this sketch so they do not need to be
//connected and should be set to -1.

#define NSS 7
#define RFBUSY 36
#define NRESET 8
#define LED1 37
#define DIO1 9
#define DIO2 -1                 //not used 
#define DIO3 -1                 //not used
#define RX_EN 21                //pin for RX enable, used on some SX1280 devices, set to -1 if not used
#define TX_EN 10                //pin for TX enable, used on some SX1280 devices, set to -1 if not used 

#define LORA_DEVICE DEVICE_SX1280                //we need to define the device we are using  

//LoRa Modem Parameters
const uint32_t Frequency = 2445000000;           //frequency of transmissions
const int32_t Offset = 0;                        //offset frequency for calibration purposes
const uint8_t Bandwidth = LORA_BW_0400;          //LoRa bandwidth
const uint8_t SpreadingFactor = LORA_SF7;        //LoRa spreading factor
const uint8_t CodeRate = LORA_CR_4_5;            //LoRa coding rate

const int8_t TXpower = 3;                       //Power for transmissions in dBm

const uint16_t packet_delay = 1000;              //mS delay between packets
```

When running this program, I get the below error:
```
ERROR - Busy Timeout!
```
The author of the StuartsProjects library has informed me that this error means the DIO1 pin on the LoRa device cannot be read, connection issue etc. As you can see in my settings, I have set DIO1 = 9 which I believe is the correct pin definition for the T3S3 v1.0 board?

Can you advise on why I might be having an issue with DIO1 and unable to use this library?

lewisxhe, 2024-03-26:
I have tested StuartsProjects/SX12XX-LoRa library before, and the test code is [here](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1280/Ranging/54_Ranging_Master). You can try it. Please note that the conclusion I have drawn from ranging is invalid so far, but StuartsProjects/SX12XX-LoRa library should work.

TomsGitBits, 2024-03-27:
Thank you, this was helpful and I was able to get the StuartsProjects/SX12XX-LoRa library working.. partially. The SD card is failing to mount. 

The example I am particularly interested in is the [233_SDfile_Transfer_Transmitter](https://github.com/StuartsProjects/SX12XX-LoRa/tree/master/examples/SX128x_examples/DataTransfer/233_SDfile_Transfer_Transmitter) program. On [line 150](https://github.com/StuartsProjects/SX12XX-LoRa/blob/3205c231f7be7fd77c9caf6167afa5f3be5838a8/examples/SX128x_examples/DataTransfer/233_SDfile_Transfer_Transmitter/233_SDfile_Transfer_Transmitter.ino#L150), the SD card is initialized using the <SdFat.h> .begin(SDCS) method. I have the pin definition for SDCS set to 13.

Any idea why this SD card issue might be appearing?

lewisxhe, 2024-03-27:
Two sets of SPI are needed. You need to initialize the SD card SPI separately and pass the SPI handle to it. Please refer to here.
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/83dd32ad95fe6a19c67065f4d33bbb72004d3360/examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt/boards.h#L196

TomsGitBits, 2024-03-27:
I am also getting the **SDCard FAILED** error on the [SX128x_Transmit_Interrupt](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/83dd32ad95fe6a19c67065f4d33bbb72004d3360/examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt/boards.h#L221) example in this library.

lewisxhe, 2024-03-27:
[233_SDfile_Transfer_Transmitter.zip](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/files/14767081/233_SDfile_Transfer_Transmitter.zip)
Give it a try, I've tested it here and it's OK

TomsGitBits, 2024-03-27:
Thanks for sending that over. SD.begin(SDCARD_CS, SDSPI) is still returning false.

After more research I believe the issue is that <SD.h> does not support SDXC cards or exFAT, but <SdFat.h> does. I see that the example you sent and the [SX128x_Transmit_Interrupt](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt) example are both using SD instead of SdFat. I think my options are:

1. Switch to an SDHC card, or
2. Get the SdFat library working so that my SDXC card can be used. Currently, SD.begin(SDCARD_CS, SDSPI) errors out if SdFat is in use.

lewisxhe, 2024-03-27:
Maybe you can try using another SD card

TomsGitBits, 2024-03-27:
I will try that. I have ordered some new SD cards.

In the meantime - I did have one other question. The [233_SDfile_Transfer_Transmitter](https://github.com/StuartsProjects/SX12XX-LoRa/tree/master/examples/SX128x_examples/DataTransfer/233_SDfile_Transfer_Transmitter) program takes files from Transmitter SD Card and sends them to Receiver SD Card. My ultimate goal is to rework this to take files from Transmitter **USB thumb drive** and send them to Receiver SD Card. 

Is it possible on this T3S3 v1.0 board to read files from a USB thumb drive? If so, are there any examples you might be able to provide?

lewisxhe, 2024-03-28:
I don't have such an example, but there are already many such examples on github, you can refer to this

https://github.com/Xinyuan-LilyGO/T-Dongle-S3/tree/main/example/USB-MSC

closed 2024-05-12 by github-actions[bot]