Xinyuan-LilyGO/LilyGo-LoRa-Series #151: About BOOT button on T-BEAM SUPREME

MrMutlu, 2024-03-27:
Hello, i want to use BOOT button on T-BEAM SUPREME as regular input button. In datasheet and header file boot button configured as GPIO0 and pulled-up internally. ( I assumed GPIO0 is boot button as like other ESP32 boards ). However when i read GPIO0 state it is always "0" and pressing BOOT button does not change its' state. Any idea why that happens?

(I don't want to use PWR button because in my code i want to use seperated button function and if its' possible doing it with BOOT button).
Thanks.

lewisxhe, 2024-03-28:
By default, a pull-up resistor has been added to GPIO0. Even if the button is damaged, the IO0 level read in the software should be HIGH. 

Use the simplest program to test the button.
```
void setup()
{
    pinMode(0, INPUT_PULLUP);
    while (1) {
        Serial.println(digitalRead(0)); delay(200);
    }
}
```

MrMutlu, 2024-03-28:
Hi, thanks for you fast response, i checked again and i am getting "0" as output. However i have second device and tried on it again ( i don't know why i didn't try that in first place ) and currently it is indeed printing "1" default and when pressed, goes to "0". So i assume my first device has a button damage but it looks like somehow is pulled to ground. I will investigate the button and try to fix it with basic soldering if its possible. 
Cheers.

closed 2024-03-28 by MrMutlu