Xinyuan-LilyGO/LilyGo-LoRa-Series #152: Lily Go T-Beam Supreme connected wtih GSM GPRS

cchantra, 2024-04-08:
Is it possible to connect Lily GO T-BEAM Supreme with  GPS GPRS  such as SIM900A ?
Can you suggest?

Regards
chantana

lewisxhe, 2024-04-10:
Of course, any free IO port can serve as Serial.
Example
```
     Serial2.begin(GPS_BAUD_RATE, SERIAL_8N1, RX_PIN, TX_PIN);
```

cchantra, 2024-04-13:
Hello,

After I put the pins on the board, my board is not working like before.
USB doesn't work anymore. After I pushed reset and plug in the usb again.
The board type before TAMC Termod S3 as in the screen below.

How to fix this? Should I reflash the firmware?

Thank you.

cchantra, 2024-04-13:
Hi
I cannot flash the firmware. The error says
Could not open /dev/cu.usbmodem1201, the port is busy or doesn't exist.

When I pressed boot and reset, the port appears but there is such error
when I tried to flash, then the port disappeared.

How to fix this?

Regards
chantana

lewisxhe, 2024-04-15:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/137

cchantra, 2024-04-19:
hello, we need to flash bootloader again with linux?

regards

cchantra, 2024-04-19:
hello, I tried to use binary from the website to flash the board. (tbeam-s3)
https://github.com/adafruit/tinyuf2/releases
but after I flashed, arduino detected the board as
 Lolin c3 mini after awhile the connected port is disconnected.

I cannot upload the program.
Is the board broken?

chantana

lewisxhe, 2024-04-22:
[image]
If you write the wrong firmware, it will reboot indefinitely and you must enter download mode manually.

closed 2024-06-05 by github-actions[bot]