Xinyuan-LilyGO/LilyGo-LoRa-Series #153: T-Beam `E (141058) task_wdt: Task watchdog got triggered.` After `enqueuing for send`

JCBuck, 2024-04-17:
I have a set of T-Beam m8n devices with board revision `T22_V1.1 20210222`
all have been flashed to `2.3.4.ea61808 Beta` with the fully erase option

I have one of 4 devices that constantly freezes/crashes right after booting

Comparing the logs side by side to one that isn't crashing or having the watchdog timeout implies that maybe the LoRa radio isn't responding... Is the radio toast? I think this was happening even before this firmware version since i got the device originally but never had time to troubleshoot it.
[image]

```haskell
DEBUG | ??:??:?? 31 [NodeInfoModule] ESP32 crypt fr=8488a5b0, num=16139130, numBytes=49!
DEBUG | ??:??:?? 31 [NodeInfoModule] enqueuing for send (id=0x16139130 fr=0xb0 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=10)
DEBUG | ??:??:?? 31 [NodeInfoModule] txGood=0,rxGood=0,rxBad=0
E (141058) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (141058) task_wdt: - loopTask (CPU 1)
E (141058) task_wdt: Tasks currently running:
E (141058) task_wdt: CPU 0: IDLE
E (141058) task_wdt: CPU 1: loopTask
E (141058) task_wdt: Aborting.

abort() was called at PC 0x4014dd54 on core 0

Backtrace: 0x400844ed:0x3ffbf44c |<-CORRUPTED
```

lewisxhe, 2024-04-22:
It seems that something blocks the running of the program, causing the watchdog to reset. Can you try this test firmware to see if each unit is normal? https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/hardware_test/t-beam-v1.x/Comprehensive
You need to write the firmware of a fixed model, and then press the IO38 button to switch the function page. After booting, diagnostic information will be output in the Serial

JCBuck, 2024-04-22:
I have 915mhz versions of the tbeam, does it matter if i flash the 868, or 433 firmware?

lewisxhe, 2024-04-22:
For the 915 version, you can only use 915 or 868. The transmission of writing 433 will cause problems.

JCBuck, 2024-04-30:
Good to know I will give the 868 a try

JCBuck, 2024-05-09:
```haskell
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
+-+-+-+-+-+-+ +-+-+-+-+-+ +-+-+ +-+-+-+-+
|L|i|l|y|G|o| |T|B|e|a|m| |S|3| |C|o|r|e|
+-+-+-+-+-+-+ +-+-+-+-+-+ +-+-+ +-+-+-+-+

-----------------------------------
Reset reason: In case of deep sleep, reset was not caused by exit from deep sleep
PSRAM is enable! PSRAM: 4.00MB
Falsh:4 MB
Falsh speed:40 M
Model:ESP32-D0WDQ6-V3
Chip Revision:3
Freq:240 MHZ
SDK Ver:v4.4.4
DATE:Jul 12 2023
TIME:16:44:34
EFUSE MAC: <redacted>
-----------------------------------
32K XTAL OSC has not started upWarning: Failed to set rtc clk to 32.768Khz !!!
========================================
========================================
setupPower
Warning: Failed to find AXP2101 power management
AXP192 PMU init succeeded, using AXP192 PMU
[ 3369][E[XPowersAXP192.tpp:1914] disablePowerOutput(): Failed to disable the power channel, the power channel has been protected
=========================================
DC1 : + Voltage: 3300 mV
DC2 : - Voltage: 1800 mV
DC3 : + Voltage: 3300 mV
LDO2 : + Voltage: 3300 mV
LDO3 : + Voltage: 3300 mV
=========================================
PowerKeyPressOffTime:4 Second
========================================
========================================
[ OLED ] Started OLED
Find display model SSD1306
========================================
========================================
setupRadio
Start Radio failed,code:-2
========================================
setupGPS
Try to init L76K
Wait L76K stop nmea timeout!
Warning: Failed to find L76K GNSS Module

Begin Ubox GNSS .
Try use 9600 baud
Module Info :
Soft version: XT CORE 3.01 (107900)
Hard version: 0080000
Extensions:7
OM BASE 2.01 (75331)
WVER=SPG 3.01
ROTVER=18.00
OD=NEO-M8N-0
IS=0xC84016 (200037)
PS;GLO;GAL;BDS
BAS;IMES;QZSS

Model:ROTVER=18.00

GetModel:NEO-M8N-0
UBlox GNSS init succeeded, using UBlox GNSS Module

========================================
========================================
========================================
Power : +
Display : +
Radio : -
GPS : +
PSRAM : +
```

The display shows
```haskell
Diagnosis
Power + OLED  +
RADIO - PSRAM +
GPS   +
```

Does this mean the Radio does not work? Could I get it replaced?

closed 2024-06-24 by github-actions[bot]