Xinyuan-LilyGO/LilyGo-LoRa-Series #156: T-Beam v1.2 GPS not working

Sailorspoon, 2024-04-21:
SOC
ESP32

Lora IC
LILYGO LORA32 868/915MHz

Description

I am trying to figure out why the GPS module is not working.

I already tested the GPS reset and test firmware and the NMEA reset. 

According to the data send to the serial monitor the GPS module can not be found.

Is this related to the AXP2101 chip of the 1.2 version or is this a hardware issue?

Here are the serial messages appearing when testing the different fixes:

GPS_reset.bin
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Connected to Serial
AXP192 Begin FAIL
All comms started

===== STATE 0`

GPS_BasicNMEARead.bin
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
initBoard
AXP192 false
SparkFun Ublox Example

Sending: CLS:CFG ID:RATE Len: 0x0 Payload:
sendCommand: Waiting for ACK response
waitForACKResponse: TIMEOUT after 1100 msec.
Ublox GPS not detected at default I2C address. Please check wiring. Freezing.`

NMEA_Reset: (The spacing is off, that is exactly the way it showed up on the serial monitor)
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
initBoard
AXP2101 PMU init succeeded, using AXP2101 PMU
=========================================
                                         DC1  : +   Voltage: 3300 mV
                                                                     DC2  : -                                                             Voltage: 0500 mV
                 DC3  : -   Voltage: 3300 mV
                                             DC4  : -   Voltage: 1840 mV
                                                                         DC5  :                                                           -   Voltage: 3300 mV
                     ALDO1: -   Voltage: 3300 mV
                                                 ALDO2: +   Voltage: 3300 mV
                                                                             ALD                                                          O3: +   Voltage: 3300 mV
                         ALDO4: -   Voltage: 3300 mV
                                                     BLDO1: -   Voltage: 3300 mV                                                          
 BLDO2: -   Voltage: 3300 mV
                             =========================================
                                                                      PowerKeyPr                                                          essOffTime:4 Second
Started OLED
recovery failed!`

lewisxhe, 2024-04-22:
Are you using the M10Q version? It seems that communication is indeed impossible. Contact the dealer for replacement.

Sailorspoon, 2024-04-22:
If you are refering to the GPS module, I don't think it is the M10Q version.
It says ublox NEO-6M-0-001 on the module.
I will contact the dealer for replacement.

closed 2024-06-06 by github-actions[bot]