Xinyuan-LilyGO/LilyGo-LoRa-Series #157: T-Beam V1.2 Mismatch Schematic to PCB - Header U12 Pin 1 5V is connected to USB VBUS0 instead of VSYS

dj9pz, 2024-04-22:
The schematic for T-Beam V1.2 does not match the board (PCB).

According to the schematic (https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/LilyGo_TBeam_V1.2.pdf) pin 1 of connector U12 (labeled "5V" on the actual board) should be connected to VSYS which is driven by the power controller AXP2101 either from VBUS0 (when powered by USB) or from the battery.

The actual PCB seems to have pin 1 of connector U12 directly connected to the USB VBUS0. When powered from USB, this pin is at 5V, when powered from 18650 battery, this pin is at 0V.

This difference between schematic and board causes problems with the Lilygo 1.3 inch OLED display, which uses pin 1 (5V) to provide power to the OLED display through a 3.3V regulator, ie. the display only works when the board is powered from USB.

lewisxhe, 2024-04-25:
Yes, it has been noticed, OLED 1.3 has been changed to DC1 for power supply

dj9pz, 2024-04-25:
Great, thanks for your reply!

Unsoldering the low power regulator and soldering a 0402 0R SMD resistor on the OLED 1.3 board would've been unfeasible for most users ;-)
[image]

lewisxhe, 2024-04-29:
Well, the LDO has been removed at the factory and the 0R resistor is connected

closed 2024-06-13 by github-actions[bot]