Xinyuan-LilyGO/LilyGo-LoRa-Series #163: T3S3 v1.2: if "USB CDC on boot" enabled, cannot boot from USB power-only?

moshe-braner, 2024-05-24:
I'm adapting a project from the old ESP32 "paxcounter" to the T3S3 (v1.2).  This is my first work on the ESP32-S3 so maybe I'm just confused.  But in what reading I have done, I could not find issue this mentioned:

I enabled "USB CDC on boot" (in the Arduino IDE) to be able to see output from Serial.print() (in the "Serial Monitor") - important for development.  That works, but if I then run the same sketch (already uploaded) on the same board with just a USB power adapter (instead of plugging into a computer) it does not finish booting.  It is as if the CDC requires handshaking with the other device (the computer)?

I have found a workaround: disabling "CDC on boot", and replacing all the Serial.* with USBSerial.* (which can be done via a #define).  I then need to press the "boot" button for each upload, but that is far better than producing a sketch that cannot run without a computer connected.  But maybe I am missing something.

moshe-braner, 2024-05-27:
It is possibly the same issue reported here:
https://github.com/espressif/arduino-esp32/issues/9545
If so, then it is a bug in the older board support Core, that has been fixed in a later version.  Unfortunately I cannot use later versions, because my compiled binary then does not fit in the flash space (and may also not have enough RAM).  With each version the Core (and other libraries) gets bigger, that is a problem.

joyel24, 2024-06-28:
I agree with the bot, @moshe-braner you can close this issue. 😉

closed 2024-06-28 by moshe-braner

michaelradhuber, 2025-01-23:
@joyel24, I think it is premature to close this issue. A lot of people are experimenting troubles with ESP32C3 (mine is a super mini F4N) not accepting external power and booting only from USB. I have the same issue, and after a lot of trial and error I found out that it WILL BOOT from the external power supply only if USB CDC is disabled on boot.

Hence I can confirm above behaviour for the ESP32 IDF version 2. I can not refer to the IDF version 3 because for a lot of legacy driver issues it won't compile, hence a lot of people are still sticking with version 2 out here. I hope it will be fixed in IDF version 3, but as long as that version remains un-usable on the Arduino IDE plattform there is no sense in closing this issue.

So for all those people spending hours and hours trying to figure out why the ESP32 C3 (or other versions) won't boot from the external power supply, try disabling USB CDC on boot and check again!

joyel24, 2025-01-23:
There is no link with T3S3 and ESP32-C3.

T3S3 is using an S3 not C3