Xinyuan-LilyGO/LilyGo-LoRa-Series #164: T3S3 USB flashing without the IDE

moshe-braner, 2024-05-24:
I'm adapting a project from the old ESP32 "paxcounter" to the T3S3 (v1.2). This is my first work on the ESP32-S3.  I use the Arduino IDE.  The "sketch uses the "minimal SPIFFS with OTA" memory model.  But the first-time flashing must be done via USB, since the board seems to arrive from the factory with Meshtastic firmware installed.  (Would be nice if we could order it with some "minimal SPIFFS with OTA" sketch with just OTA ready to use...)

Eventually I will offer this open-source program to others, who will want to first-time-flash it to their boards without having to set up the development environment.  For the ESP32 version the instructions were to download the standalone ESPtool, and to set it up to flash the main .bin plus 3 other small .bin files (boot_app, bootloader, partitions).  Can one use these same additional files for the ESP32-S3, or are they target-specific?  If so, where can I get the equivalent files for the S3?

lewisxhe, 2024-05-25:
The partition table provided by Arduino IDE is limited. The best way is to use platformio to customize the partition table.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html

moshe-braner, 2024-05-27:
Thank you.  But the partition table is OK for my purpose.  And the question, how people receiving my compiled binary can flash it onto their device without setting up any IDE, does not depend on what IDE I used to compile it.  I have found a way though: in the Arduino IDE, in the preferences, set "upload" to "verbose".  Then ask it to upload (without even having a board connected, it will fail, that is OK).  Before uploading it will then show the long command line being passed to the ESPtool.  Within that line you can see the pathnames of the uploaded files, preceded by the flash addresses each file it directed to.  Go to those paths and copy the files elsewhere.  The boot_app0 and partitions .bin files are identical to those used with the ESP32, but the bootloader.bin is different for the S3.

closed 2024-07-11 by github-actions[bot]