Xinyuan-LilyGO/LilyGo-LoRa-Series #165: NEO M8N disarmed

dms-rj, 2024-05-24:
I would like to receive a step by step method to restore or reset the NEO M8N  of the T BEAM AXP2101 with SX1262 and ESP32-DOWDQ6 V3. I've tried a lot of simple solutions such as simple reset by RST button of the own radio and reset firmware on the flasher.mashtastic.org and I received the following images, the issue was provoqued due to an installation of a display OLED 1.3 with 3 buttons, it cames arrives with an issue that was resolved with a jumper oriented by the Lilygo store, solution that worked in two devices but not on other two (I have four), and did red light up and turned off the GPS function.
[image]
[image]
[image]
[image]
[image]

lewisxhe, 2024-05-25:
Try the method here to see if the GPS can be reset
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/GPS_Reset

dms-rj, 2024-05-29:
> Try the method here to see if the GPS can be reset https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/GPS_Reset

I',ve tried that but dind't work. I guessed that I killed the T-Beam with it procedure (i am a noobie in electronic), so I've tried a solution of the Lilygo engineers and all the T-Beam turned back. They sent me a ESP32 app and a recovery file that I recharged and reseted the T-Beam. Than I recharged the T-Beam firmware version and now I am testing it normaly.

dms-rj, 2024-05-29:
The case was solved, and the explanation is in the reply of the help solution that, with me (repeat, with me), didn't work, thank you.

closed 2024-05-29 by dms-rj