Xinyuan-LilyGO/LilyGo-LoRa-Series #168: SD with LoRa fails

mkarat99, 2024-05-30:
I'm using T-Beam Supreme (M) board and trying to use the SD card with LoRa but it seems that there is a conflict to the SPI initialization. While the SD.begin() is successful and appends data on the SD card file, once LoRa transmission is executed (and vice versa), the board crashes. Do you have any suggestion for a solution to this?

Thank you!

lewisxhe, 2024-06-03:
Refer to this example
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples

lewisxhe, 2024-06-21:
Can you post your test code?

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/ba3fda6c06d44d7f95fe842b8f7fb0f7e68f89b7/examples/RadioLibExamples/Transmit_Interrupt/LoRaBoards.cpp#L324

closed 2024-08-06 by github-actions[bot]