Xinyuan-LilyGO/LilyGo-LoRa-Series #170: T3-S3 SX1262 will not boot with any powerbank

madler, 2024-06-06:
When get USB-C power from my desktop or laptop computer, my LILYGO LoRa T3-S3 works great. The blue LED comes on, it boots, and it operates, flashing the green LED.

However if I try to plug it into any of my powerbanks, the blue LED comes on, but it refuses to boot. This is on three different powerbanks with USB-C supplies, from different manufacturers including Anker, as well as two very large batteries (Jackery and Bluetti), also with USB-C supplies.

I have a USB-C power monitor, which shows 5+ V being supplied in all cases, powerbank or computer, with a few ten's of mA draw.

Why is this radio refusing to work with powerbanks?

moshe-braner, 2024-06-06:
See my related thread: https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues/163
After I switched to the ESP board support "core" from version 2.0.3 to 2.0.17 this problem was solved.  It boots OK from a plain USB power source, no matter whether I compile with "USB CDC on boot" enabled or disabled.  This was supposedly a bug in Serial.print() in the older versions of the Arduino-ESP board support.

madler, 2024-06-07:
Thanks! I updated the software to an Alpha version and that must have the updated core code since it now works with powerbanks.

closed 2024-06-07 by madler

oshanz, 2024-06-29:
Hi, does this board support PD negotiation. or do we need to supply fixed 5v from usb c?

moshe-braner, 2024-06-29:
If you plug a dumb (no PD negotiation) device into a smart PD USB power source, doesn't the power source default to 5V?

madler, 2024-06-29:
Yes, it always starts with 5V until negotiated otherwise.

oshanz, 2024-06-29:
Awesome! didn't knew how type c PD operate. Thank you both.