Xinyuan-LilyGO/LilyGo-LoRa-Series #171: WiFi AP "password incorrect" in one T3S3 board

moshe-braner, 2024-06-06:
I cannot connect to the wifi network of one of my T3S3 boards (out of 3) when it is in SoftAP mode.  The same .bin flashed into another board of same model works without problems.  Also, the same board that does not connect in SoftAP mode connects with no problems if set instead to be a client on the house wifi network.  How can this happen?  Is the hardware in this specific board defective?  In a way that is specific to the wifi encryption in SoftAP mode?  Or is there a bug in a library, that is specific to this board?  For example, perhaps the encryption algorithm uses the chip ID or MAC address in a way that can fail on some boards and not others?

lewisxhe, 2024-06-12:
Is the AP generated normally? If the AP hotspot is generated normally, use the simplest data flow to test whether the web page can be opened.

lewisxhe, 2024-06-12:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino

Use this example to test, and note that the default account and password need to be changed.
```
// Set these to your desired credentials.
const char *ssid = "yourAP";
const char *password = "yourPassword";
```

moshe-braner, 2024-06-12:
Thanks Lewis.  But I don't understand how trying to access the web page would help.  I cannot connect to the wifi network itself (the AP), it tells me "password incorrect".  But the password is correct.  The exact same compiled binary, flashed onto another T3S3 board, works correctly - I can connect to the AP network it creates, and then the rest of the program runs normally.  Even the board that does not allow the connection runs normally otherwise, and after I load the house wifi SSID and password into it (via the SD card) then it connects to the house wifi with no problems.   Only the initial WiFi AP fails with the "password incorrect" message, and only on this board, not the other identical board with the identical binary.

lewisxhe, 2024-06-12:
Maybe you can try to erase the internal data of the flash and then test it again.

[tools](https://www.espressif.com.cn/sites/default/files/tools/flash_download_tool_3.9.6_2.zip)

[image]

lyusupov, 2024-06-18:
[image]

[image]

moshe-braner, 2024-06-26:
I finally remembered to do the test, used the ESPtool to "erase the flash", and that fixed the problem: after I uploaded my program again (also via the ESPtool with the right auxiliary files) it was possible to connect to the WiFi AP it created, with the password that is hardcoded into the program.  BTW (lyusupov) I compiled this program with ESP Core 2.0.17, which I had to shift to since older versions (prior to 2.0.15) have a bug regarding the S3 that prevents "Serial" output via USB and/or hangs on Serial output (appears to not boot), depending on the "USB CDC on boot" setting.

Strangely, the two other T3S3 boards that I bought at the same time with the same software (Meshtastic) installed at the factory did not have this problem.

lyusupov, 2024-06-26:
[image]

moshe-braner, 2024-07-22:
There is still a problem.  I bought 3 identical T3S3 boards at the same time with the same Meshtastic software pre-installed.  I put my own software on them, replacing Meshtastic.  The board I could not connect to its AP, was fixed after a full erase of the flash, as noted before.  I thought the second one was OK, but apparently not - same problem but also the same solution.  OK.  But the third one, has a different but related problem: the AP SSID does not appear in a phone or tablet that lists the available networks, even though my (same) software displays the SSID on the OLED and says it's ready.  I tried a flash erase, and that did not fix this problem on this board.  That same board works OK (can connect to the web server it creates) when my software connects to an external network instead of creating its own.

moshe-braner, 2024-07-28:
Took that device home and now the wifi network it creates does appear in other devices.  Unlike how it behaved in the field.  I am confused.

moshe-braner, 2024-08-02:
Wild conjecture:  The weather was hot.  The device was mounted at the bottom of a dark box with a solar panel on top.  Maybe the ESP32S3 refuses to turn on the WiFi transmitter if it is hot?

lewisxhe, 2024-08-14:
I can't answer this for you. From the espressif manual, I see that the maximum working temperature is within 125°, but I'm not sure. Maybe you can test it by changing the environment to see if it is the temperature that affects it.

closed 2024-09-28 by github-actions[bot]