Xinyuan-LilyGO/LilyGo-LoRa-Series #172: T3S3 LoRa Board - Stuck in Bootloader

starlight-traveler, 2024-06-11:
Hello,

I have attempted to flash the T3S3 LilyGo LoRa board with a custom program, and while it says it has successfully flashed, when I open serial the board is stuck in:

```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x22 (DOWNLOAD(USB/UART0))
Saved PC:0x40045ab3
waiting for download
```

Could anyone provide guidance? From another issue it seems like this is hardware related, but not sure: https://github.com/mu-editor/mu/issues/2313

baisch, 2024-06-11:
Same issue on my custom board. GPIO0 is connected to a switch (change boot to flash-mode). But I always get:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x22 (DOWNLOAD(USB/UART0))
waiting for download

any ideas to fixed it?

lewisxhe, 2024-06-12:
Disconnect USB and power on again, or press RST to exit download mode

starlight-traveler, 2024-06-12:
Wish that was the simple fix, but that does nothing. Regardless of whether the board is rebooted or what not it is stuck in download mode.

This has to be due to something going on at the broken boot loader level but I do not know what it is.

lewisxhe, 2024-06-12:
Did you connect anything externally to the board?

baisch, 2024-06-12:
What does boot:0x22 mean?
https://docs.espressif.com/projects/esptool/en/latest/esp32s3/advanced-topics/boot-mode-selection.html

closed 2024-07-28 by github-actions[bot]