Xinyuan-LilyGO/LilyGo-LoRa-Series #173: RadioLib PingPong not working

vitalmuffin, 2024-06-11:
I could need some help with LoRa communication between an T-Beam SoftRF SX1626 and a T-Beam Mashtastic. I tried to use the transmitter and receiver code from:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples

but this didn't work.

then i used the base of this code (utilities.h, LoRaBoards.h, LoRaBoards.cpp) and changed the main.ino to this:

`/*
   RadioLib Transmit with Interrupts Example

   This example transmits packets using SX1276/SX1278/SX1262/SX1268/SX1280/LR1121 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

#include "LoRaBoards.h"
#include <RadioLib.h>

// save transmission states between loops
int transmissionState = RADIOLIB_ERR_NONE;

// flag to indicate transmission or reception state
bool transmitFlag = false;

// flag to indicate that a packet was sent or received
volatile bool operationDone = false;

#if     defined(USING_SX1276)
#define CONFIG_RADIO_FREQ           868.0
#define CONFIG_RADIO_OUTPUT_POWER   15
#define CONFIG_RADIO_BW             125.0
SX1276 radio = new Module(RADIO_CS_PIN, RADIO_DIO0_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

#elif   defined(USING_SX1278)
#define CONFIG_RADIO_FREQ           433.0
#define CONFIG_RADIO_OUTPUT_POWER   17
#define CONFIG_RADIO_BW             125.0
SX1278 radio = new Module(RADIO_CS_PIN, RADIO_DIO0_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

#elif   defined(USING_SX1262)
#define CONFIG_RADIO_FREQ           868.0
#define CONFIG_RADIO_OUTPUT_POWER   22
#define CONFIG_RADIO_BW             125.0

SX1262 radio = new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

#elif   defined(USING_SX1280)
#define CONFIG_RADIO_FREQ           2400.0
#define CONFIG_RADIO_OUTPUT_POWER   13
#define CONFIG_RADIO_BW             203.125
#ifdef T3_S3_V1_2_SX1280_PA
// PA Version power range : -18 ~ 3dBm
#undef CONFIG_RADIO_OUTPUT_POWER
#define CONFIG_RADIO_OUTPUT_POWER  3
#endif
SX1280 radio = new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

#elif   defined(USING_SX1268)
#define CONFIG_RADIO_FREQ           433.0
#define CONFIG_RADIO_OUTPUT_POWER   22
#define CONFIG_RADIO_BW             125.0
SX1268 radio = new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);

#elif   defined(USING_LR1121)
#define CONFIG_RADIO_FREQ           868.0
[... 1429 lines cut ...]

        u8g2->sendBuffer();

    }
#endif
}

static uint8_t ledState = LOW;
static const uint32_t debounceDelay = 50;
static uint32_t lastDebounceTime = 0;

void flashLed()
{
#ifdef BOARD_LED
    if ((millis() - lastDebounceTime) > debounceDelay) {
        ledState = !ledState;
        if (ledState) {
            digitalWrite(BOARD_LED, LED_ON);
        } else {
            digitalWrite(BOARD_LED, !LED_ON);
        }
        lastDebounceTime = millis();
    }
#endif
}`

lewisxhe, 2024-06-12:
Do you have SX1262 modules on both boards?

vitalmuffin, 2024-06-12:
> [Verfolgungsbild][https://github.com/notifications/beacon/ALWXXU23MGGWYF3BSHSVCULZG6S23A5CNFSM6AAAAABJFEWNMKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUA3SL7K.gif]

no on one board it's sx1262 and on the other sx1276.

lewisxhe, 2024-06-12:
Try adjusting CR and SF in Receive_Interrupt and Transmit_Interrupt to 7.
[image]

vitalmuffin, 2024-06-12:
This worked! Thx alot. Can you explain why this works? 
And maybe you can also tell me which settings work best for long range communication (1-5 km)? I use the original antenna which came with the device. Packet size will be max 256 bytes.

lewisxhe, 2024-06-21:
SX1262 uses TCXO crystal oscillator
SX1276 uses ordinary crystal oscillator, some parameters are not common
As for the communication distance, it is adjusted according to the actual situation. I can't give you a definite parameter. The larger the data packet, the longer the transmission time. You should actually test it.

closed 2024-08-06 by github-actions[bot]