Xinyuan-LilyGO/LilyGo-LoRa-Series #176: Unable to select the respective board during flash procedure

strobesactual, 2024-06-20:
I plug the Lilygo board into my Mac and i cannot see it on the left PlatformIO menu
[image]

I am following the instructions listed on: https://www.lora-aprs.info/docs/LoRa_APRS_iGate/quick-start-guide/
I successfully get to Flashing, but am unable to select the device.
I am using a LILYGO T3 v1.6.1 Model LORA32

To Reproduce
Follow the link and go to "Flashing the Configuration" step 2

Expected behavior
I should see the board

OS: Mac Sonoma 14.5
PlatformIO Version: 3.3.3
Firmware Version:
Board name:

LILYGO LoRa32 433Mhz ESP32 Development Board OLED 0.96 Inch SD Card BLE WiFi TTGO Paxcounter Module
https://a.co/d/0dd9yFzB

lewisxhe, 2024-06-21:
Have you tried using a USB-A to USB-C cable? Have you installed the CH34x driver?
http://www.wch.cn/downloads/CH34XSER_MAC_ZIP.html

strobesactual, 2024-06-21:
Lewis,

Hello I installed the drivers and have used a USB-C to Micro USB cable. I
see the drive on the computer but its not showing up on VS code to allow me
to work on it.
[image: image.png]
[image: image.png]
I believe this is correct but nothing shows up on VS Code:
[image: image.png]
This is the device:
[image: IMG_0717 Large.jpeg]

Thank you for your help!

Kindly,
Austin

strobesactual, 2024-06-21:
I reinstalled the drivers on a separate computer and it worked. I think the first computer was having issues.

closed 2024-06-21 by strobesactual