Xinyuan-LilyGO/LilyGo-LoRa-Series #177: SoftRF Prime MK3 not receiving GPS Position / Satellite Connection

LouisTrinczek, 2024-06-21:
Hello,

I already had opened an issue at the SoftRF Github Repository, but they sent me to you, the manufacturers of the LilyGo ESP32-S3.

https://github.com/lyusupov/SoftRF/issues/169

As stated in this Issue, I am not able, when running the latest SoftRF Firmware, to connect to any GPS Satellites and receive GPS Data.

How would I be able to troubleshoot this issue?

LouisTrinczek, 2024-06-21:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/GPS_Reset
I installed all of these firmwares one after another, but it didn't seem to fix it as well.

I can connect in u-center2, but it won't show me any gps data

lyusupov, 2024-06-22:
[image]

<br>
<br>

[image]

LouisTrinczek, 2024-06-22:
Hey @lyusupov, thanks for your reply.

I left it out waiting for sometimes 10-15 minutes.

And on top of that, I don't use the antenna that is shipped with it, I ordered the larger antenna that is listed in the SoftRF Docs and screwed it on.

So that's why i'm still wondering why it won't receive any signals.

Just wondering, so that it isn't a factor: I don't have the 18650 battery installed yet, but i'm powering it with USB from the laptop. Shouldn't be a problem right?

lyusupov, 2024-06-22:
>  ordered the larger antenna that is listed in the SoftRF Docs

there are no no any **GNSS antenna** "listed in the SoftRF Docs"
https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkIII#bill-of-materials

The SoftRF Gitter thread is talking about **active** (built-in LNA) **GNSS antenna** with uFL connector that can substitute the LilyGO stock one if it is defective or does not supply enough gain.

> i'm powering it with USB from the laptop

USB power noise can kill sensitivity of any ISM band radio and GNSS modules.

LouisTrinczek, 2024-06-22:
What is this larger 868 MHz Antenna that I put on? Is it not also for better GPS Receiving?
https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkIII#recommended

LouisTrinczek, 2024-06-22:
[image]
This has been lying outside for 35 Minutes now again

LouisTrinczek, 2024-06-22:
I left it out for a bit longer now. 

After 4 and a half hours it somewhen began transmitting and receiving signals, also showing a gps Position. But still no gps satellites, which doesn't quite make sense, the height is wrong and the position is only accurate to ~ 4km. 

That is usually not how it's supposed to be, right?

[image]

lewisxhe, 2024-06-26:
Are you using Ublox or L76K? If the antenna is in the visible sky, the positioning speed of these two modules is generally 1-2 minutes, and in a clear sky, it can be locked in one minute. I think the antenna problem should be ruled out first.

LouisTrinczek, 2024-07-04:
After leaving it out once for a few more hours, it then started working apparently.

Was just weird because the initial setup time was way too long for me.

But for subsequent starts it worked better.

Don't have any more issues with it as of right now.
Thanks

closed 2024-07-04 by LouisTrinczek