Xinyuan-LilyGO/LilyGo-LoRa-Series #178: T3S3 v1.2: No USB Power Output enen in USB OTG Host mode (Power via Battery JST only)

joyel24, 2024-06-22:
I would like to use the T3-S3 USB-C port as a USB OTG Host.
Data+/Data- appears to have proper communication but +5V is outputting 0V so it cannot power anything...

Is there a simple way to enable +5V output on USB-C ?

I searched a bit in the pcb, for the moment, the best option I found is to solder a little wire from [5Vpin or ON/OFF SW] to this side of this tiny 10k resistor witch appears to be connected to the USBC 5V port:

BUT is there a better way to have +5V on USB-C port, or is there a better place to easier solder a wire 

[image]

lewisxhe, 2024-06-26:
If you do that, the battery voltage will be directed to USB-C. If you plug the USB-C into the computer at this time, it is equivalent to connecting the battery to 5V. This is dangerous behavior. In addition, the battery can only provide a maximum voltage of 4.2V (when fully charged) and cannot provide a standard 5V for USB devices. I do not recommend using it this way.

joyel24, 2024-06-26:
[image]

moshe-braner, 2024-06-27:
I wish the LilyGo boards (T3S3, T-Beam, etc) would have a solder terminal that connects to the 5V USB power.  For purposes such as this.  Or in general to be able to supply 5V power to the board without using a USB connector.  For example, I have powered some T-Beams from a 12V-5V converter that does not have a USB connector, how to attach it to the T-Beam is a problem.  Also, the existing solder terminal that is marked "5V" should not be marked that way, since that is not what it is!

joyel24, 2024-06-27:
> I wish the LilyGo boards (T3S3, T-Beam, etc) would have a solder terminal that connects to the 5V USB power. For purposes such as this. Or in general to be able to supply 5V power to the board [...]

I agree, I think there is a bit of improvements to power an USB device to it while running on battery.
It could be a little DC-DC converter switchable like in the flipper zero for example. You can enable 5V output when you want but it's not enabled by default to draw less power probably.

moshe-braner, 2024-06-27:
Powering an external USB device while running on battery would, as you said, require additional hardware, another DC-DC converter.  That may or may not be not worth adding to the basic board, increasing its cost for a rare type of usage.

What I requested was just a solder terminal on the board, that is connected directly to the power pin of the existing USB jack.  That would cost nothing, just a small change in the design of the PCB.

For the purpose of powering an external USB device, it could then be done by using an external 5V power supply through that added solder terminal.  That by itself would not allow extended standalone battery-powered operation of the pair of devices, though.

I have set up a solar-powered system with a T3S3, a small solar panel, and an external DC-DC converter the takes the solar panel output and converts to 5V which is fed to the USB jack (and it would have been easier if there was a true 5V solder pad).  A USB OTG device could be powered by that either by a connection to the solar 5V source (can use an OTG Y cable) or by removing the battery from the OTG device and powering it instead from the solder tab currently marked "5V" - which is really about 4V - connected to the battery terminal (not the USB jack) of the external device.

joyel24, 2024-06-28:
Yes, just an available pin connected to 5V of USB port can be enough. It can be anywhere in the board but i'll be way better than doing like I did.

joyel24, 2024-12-03:
@Xinyuan-LilyGO Did you read the end of this issue?

lewisxhe, 2024-12-03:
Hi  @joyel24 

Sorry, I overlooked that.

You want to power from the 5V pin and then output 5V from USB-C, right?

If so, you may need to remove the charging chip, otherwise the input and output of the charging chip will be shorted together. At this time, battery power is unavailable.

If you are sure you need to do this, I can ask how to change it.

joyel24, 2024-12-03:
If you just append a pin connected with 5V USBC port.

It can solve the trouble.

When using the board under battery, there is no way to connect any USB device as it won't be powered. 

Voltage coming from bettery is enough to power most of USB peripherals.

So if we have something like Vbat and 5VUSB both accessible in the board, we can short the charging circuit to power USB OTG easily without needing Y cable or any other annoying pcb modification...

Ex:
Connection an usb keyboard with something like meshtastic firmware...

lewisxhe, 2024-12-03:
I understand what you mean, but it cannot be done under the current hardware. It can only be done by short-circuiting D13, right!
[image]

joyel24, 2024-12-03:
> I understand what you mean, but it cannot be done under the current hardware. It can only be done by short-circuiting D13, right!
> [image]
> 

That's exactly what I was thinking about.
This is the mod I made, and I confirm it's working. 

 but it will be better for everyone can simply have access to

joyel24, 2024-12-03:
> I understand what you mean. These changes can only be revised in the next version.
> Is it like the diagram below?
> [image]
> 

100%

Like in the comment of this issue. 
[image]

joyel24, 2024-12-03:
But we need both pins to be honest, not only 1 of them. 

Then with the extra VBUS and VBAT pins accessible in the board, we can simply short them together when needed and it will be very useful.

lewisxhe, 2024-12-03:
so ?
[image]

joyel24, 2024-12-03:
But we need both pins to be honest, not only 1 of them. 

Then with the extra VBUS and VBAT pins accessible in the board, we can simply short them together when needed and it will be very useful.

1000% 🥳

lewisxhe, 2024-12-03:
So, that's it ??? Right?
If so, I'll give feedback and submit revisions right away

joyel24, 2024-12-03:
> So, that's it ??? Right?
> If so, I'll give feedback and submit revisions right away

Sure, don't forget to submit your marketing service to send me a T-deck Plus. I missed black Friday promo after some hours... I even sent an email for that because the code wasn't working. 

I really want to participate in the tft ui development to run meshtastic on it... 

I believe it's possible to have a free sample unit if you ask ?

lewisxhe, 2024-12-03:
1. I believe it's possible to have a free sample unit if you ask ?

> Of course, that's what I want to say. I can apply for a free sample from you.

2. Don't forget to submit your marketing service to send me a T-deck Plus. I missed black Friday promo after some hours... I even sent an email for that because the code wasn't working.

> Did you encounter any problems with T-Deck-Plus ? Did you open an issue or contact the dealer ?

joyel24, 2024-12-03:
No, I have no trouble with tdeck, and about the promo code it was just too late.

For the moment I never owned Tdeck+

But I realized that there is a lot of work in the ui part and I really wanted to help for it.

Is there a specific process to do for the T-deck+ free sampe ?

joyel24, 2024-12-03:
Right, the good news is that they will be optional and backward compatible if we don't solder them.

We just need to move the RST button.

lewisxhe, 2024-12-03:
Do you want a free T-Deck+ sample?

Regarding the T3-S3-V1.3 revision, I added some of my own opinions, canceled the design of soldering OLED, changed to FPC socket for connecting OLED, and moved the RST button to the same side as the BOOT button.

lewisxhe, 2024-12-03:
Give me your email address, I will request a sample for you, and then a salesperson will contact you via email.

joyel24, 2024-12-03:
> Give me your email address, I will request a sample for you, and then a salesperson will contact you via email.

So nice, there it is, this one is coming from Google Mail...
joel.martn.2104[...]

lewisxhe, 2024-12-03:
Huh? Give me your email address.

joyel24, 2024-12-03:
Don't worry it's mine, I even already placed orders to lylygo.cc with this email...
The end is gmail as you can imagine.

lewisxhe, 2024-12-03:
OK, you mean you've sent this link to our sales people? OK, that's fine.

joyel24, 2024-12-03:
No I said it's a real mail that I own and with this one I already bought items from the lilygo shop.

No I didn't contact them

lewisxhe, 2024-12-03:
Then you should send me your email address. Otherwise I don't know how to send you the sample.

joyel24, 2024-12-03:
I just sent you an empty mail

lewisxhe, 2024-12-03:
Got it.

closed 2024-12-03 by lewisxhe

joyel24, 2024-12-03:
Thanks to you. ♥️
I hope to see this future version in the market very soon 😊

joyel24, 2025-03-22:
Nice ! I see that it's just released  <3 Great !