Xinyuan-LilyGO/LilyGo-LoRa-Series #179: T3S3: Add support to Adruino-esp32

joyel24, 2024-06-24:
Please add T3S3 to the supported boards for Arduino-esp32.
I've open an issue here: https://github.com/espressif/arduino-esp32/issues/9927

lewisxhe, 2024-06-26:
T3-S3 has many variants. A single version is only suitable for one variant and cannot be compatible with all. Using ESP32S3 Dev Module can be more flexible.

joyel24, 2024-06-28:
> T3-S3 has many variants. A single version is only suitable for one variant and cannot be compatible with all. Using ESP32S3 Dev Module can be more flexible.

I'm not really alright with this answer. For example T-Beam has also many variants but is available in the esp32-arduino boards list with all variants...

[image]

lewisxhe, 2024-06-28:
Ok, you found it, 🤣, I'll find time to update T3

lyusupov, 2024-07-03:
@joyel24  

https://github.com/espressif/arduino-esp32/pull/9956

[image]

joyel24, 2024-07-03:
https://github.com/espressif/arduino-esp32/issues/9927#issuecomment-2207040234

lyusupov, 2024-07-17:
@joyel24 
> ... I wanted to wait for the next release ...  before closing the issue...

[image]

closed 2024-08-31 by github-actions[bot]