Xinyuan-LilyGO/LilyGo-LoRa-Series #180: LoRa32 V2.1_1.6

pbh8075, 2024-06-26:
In the pin diagram, number 27 of mosi, number 5 of sclk, and number 18 of cs are missing. We are suffering because we cannot resolve this. please help
[image]

lewisxhe, 2024-06-27:
SPI is not reserved on both sides of the board

closed 2024-08-11 by github-actions[bot]

agieocean, 2024-11-02:
> SPI is not reserved on both sides of the board

Does this mean you'd need to disable the screen to use the micro sd? Or can you not use the micro sd and the antenna at the same time at all?

Im trying to use the micro sd slot and have not been able to find the proper pin diagram for this board.