Xinyuan-LilyGO/LilyGo-LoRa-Series #182: Unable to Read Sensor Data via I2C on LilyGO LoRa32

Mariiian09, 2024-06-27:
I am experiencing issues with connecting a sensor to the I2C interface on my LilyGO LoRa32 board. I have connected the sensor (SHT20 humidity and temperature DFRobot) to pins 21 (SDA) and 22 (SCL) but I am not receiving any data from the sensor. The sensor has four pins: VCC, GND, SDA, and SCL.

lewisxhe, 2024-06-28:
Please specify the board model or provide a link

Mariiian09, 2024-06-28:
> Please specify the board model or provide a link

The model is T3_V1 6.1 https://www.lilygo.cc/products/lora3

lewisxhe, 2024-06-29:
Paste your test sketch

Mariiian09, 2024-06-29:
> Paste your test sketch

#include <Wire.h>
#include "DFRobot_SHT20.h"

DFRobot_SHT20 sht20(&Wire, SHT20_I2C_ADDR);

void setup() {
  Serial.begin(115200);

  // Inicializa la comunicación I2C con los pines GPIO21 (SDA) y GPIO22 (SCL)
  Wire.begin(21, 22);

  // Inicializa el sensor SHT20
  sht20.initSHT20();
  delay(100);
  Serial.println("Sensor init finish!");

  // Verifica el estado actual del SHT20
  sht20.checkSHT20();
}

void loop() {
  // Lee la humedad medida por el sensor
  float humd = sht20.readHumidity();

  // Lee la temperatura medida por el sensor
  float temp = sht20.readTemperature();

  Serial.print("Time:");
  Serial.print(millis());  
  Serial.print(" Temperature:");
  Serial.print(temp, 1);   
  Serial.print("C");
  Serial.print(" Humidity:");
  Serial.print(humd, 1);    
  Serial.print("%");
  Serial.println();

  delay(1000);
}

closed 2024-08-13 by github-actions[bot]