Xinyuan-LilyGO/LilyGo-LoRa-Series #183: T-Beam SupremeS3 Schematic: Connection of RTC_INT to PCF8563 pin#3 makes no sense, is it wrong?

cyberman54, 2024-07-01:
[Schematic in this repo](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/LilyGo_T-BeamS3Supreme.pdf) shows RTC chip interrupt output (RTC_INT) connected to pin#3 od PCF8563. This makes no sense, since pin#3 ist I2C address pin (A0). RTC_INT should be connected to pin#7 instead.

Is the schematic wrong, or is it wrong on the board?

lyusupov, 2024-07-03:
There is a variety of packages for PCF8563...
Some of them are using pin#3 for the purpose of INT.

[image]

One can verify proper operation of the RTC IC by executing the sketch: https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/Sensor/PCF8563_SimpleTime

lyusupov, 2024-07-03:
[image]