Xinyuan-LilyGO/LilyGo-LoRa-Series #186: LoraSender issues while running the example code

dewaste08, 2024-07-03:
Hi there, 

I am trying to use the LoRa transmitter, i wanted to try this by running the example code "LoRaSender" but i constantly have this message on the serial monitor : 

`
-----------------------------------
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1288
load:0x40078000,len:13872
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3048
entry 0x40080590
setupBoards
-----------------------------------
Reset reason: In case of deep sleep, reset was not caused by exit from deep sleep
PSRAM is disable!
Flash:4 MB
Flash speed:80 M
Model:ESP32-PICO-D4
Chip Revision:101
Freq:240 MHZ
SDK Ver:v5.1.4-358-gbd2b9390ef-dirty
DATE:Jul  3 2024
TIME:22:12:30
EFUSE MAC: D8CA061C0610`

How can i solve this issue ? 

Thank you very much !

lewisxhe, 2024-07-04:
Did you add any code? I tested that the watchdog reset does not occur. This is usually caused by the thread being blocked.

dewaste08, 2024-07-04:
No it was the example program, without any modifications :
```C
// Only supports SX1276/SX1278
#include <LoRa.h>
#include "LoRaBoards.h"

int counter = 0;

void setup()
{
    setupBoards();
    // When the power is turned on, a delay is required.
    delay(1500);

#ifdef  RADIO_TCXO_ENABLE
    pinMode(RADIO_TCXO_ENABLE, OUTPUT);
    digitalWrite(RADIO_TCXO_ENABLE, HIGH);
#endif

    Serial.println("LoRa Sender");
    LoRa.setPins(RADIO_CS_PIN, RADIO_RST_PIN, RADIO_DIO0_PIN);
    if (!LoRa.begin(LORA_FREQ_CONFIG)) {
        Serial.println("Starting LoRa failed!");
        while (1);
    }
}

void loop()
{
    Serial.print("Sending packet: ");
    Serial.println(counter);

    // send packet
    LoRa.beginPacket();
    LoRa.print("hello ");
    LoRa.print(counter);
    LoRa.endPacket();

    if (u8g2) {
        char buf[256];
        u8g2->clearBuffer();
        u8g2->drawStr(0, 12, "Transmitting: OK!");
        snprintf(buf, sizeof(buf), "Sending: %d", counter);
        u8g2->drawStr(0, 30, buf);
        u8g2->sendBuffer();
    }
    counter++;
    delay(5000);
}
```

MoesMo, 2024-07-04:
down version 3.0.0 esp32 espressif systems\  board -arduino ide

closed 2024-08-19 by github-actions[bot]