Xinyuan-LilyGO/LilyGo-LoRa-Series #187: RSSI signal not good

cmussoni, 2024-07-04:
Good morning,
I have two "T3 V1.6.1" boards (ESP32 433/868/915MHZ 0.96 Inch OLED SD Card WIFI Module[Q134]), and used the ArduinoLoRa Transmitter and Receiver examples. But the signal, even if the two Esp32 are close, is very bad. What could it be? Thanks for a possible answer.
Claudio

lewisxhe, 2024-07-04:
Try RadioLib, ArduinoLoRa is the default parameter, it does not set the maximum power
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples

cmussoni, 2024-07-04:
> Try RadioLib, ArduinoLoRa is the default parameter, it does not set the maximum power https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples

Perfect, now I have an RSSI signal between -5 and -8 dBm.
Thanks
Claudio

closed 2024-08-19 by github-actions[bot]