Xinyuan-LilyGO/LilyGo-LoRa-Series #190: DCDC1 & DCDC3 on T-Beam v1.2

moshe-braner, 2024-07-22:
On the T-Beam v1.1 the ESP32 runs off of "+3V3" which is connected to "VDD3V3" which comes from the DCDC3 output of the AXP192.  But in the schematic for the v1.2 it seems that the ESP32, running off of VDD3V3, is connected to DCDC1 (pins LX1 and FB1) on the AXP2101.  Different from the v1.1, or an error in the schematic?  I need to know because I want to turn off the other power channel (the "3.3V" solder pads, e.g., the OLED, and other added peripherals) at some times.  But I don't want to accidentally turn off the ESP32.

lewisxhe, 2024-08-03:
The screen and AXP share the I2C bus, and the OLED power cannot be turned off. If you turn off the OLED power, the I2C bus will not be accessible normally, causing the device to fail to turn on.

moshe-braner, 2024-08-03:
Thank you.  So presumably that is also true for the v1.1?  We still don't know which of the two DCDC converters runs what on the v1.2, but you say that has no practical effect.

lewisxhe, 2024-08-14:
v1.2 has only one DC1, the others are not used

moshe-braner, 2024-08-14:
But the v1.2 schematic shows (at the AXP2101 section):
* pins  21,22 (presumably (DCDC1) connecting to "VDD3V3"
* pins 4,5 (presumably DCDC3) connecting to "VCC_2_5V"
VDD3V3 is connected to "+3V3" which is connected to multiple pins of the ESP32, so that DCDC1 is powering the ESP32.  VCC_2_5V - that is DCDC3 - is connected to the "3.3V" pins in the solder headers.

On the T-Beam v1.1 shows "+3V3" connected to "VDD3V3" instead of "VCC_2_5V", but "VDD3V3" comes from DCDC3 - pin 17 on the AXP192.  And "VCC_2.5V" comes from DCDC1 (pin 47 of the AXP192).   "VCC_2_5V" is connected to the solder header pins.  The ESP32 is powered by "+3V3", that is, from DCDC3, and the headers from DCDC1.

So according to the schematics DC1 & DC3 are reversed between v1.1 and v1.2?  Unless something is not right in the pin labeling?  This datasheet says pins 21,22 are indeed DCDC1:  https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/K128%20CoreS3/AXP2101_Datasheet_V1.0_en.pdf

It is therefore not clear what would happen if my code would do this:
   PMU->disablePowerOutput(XPOWERS_DCDC1);
or this:
   PMU->disablePowerOutput(XPOWERS_DCDC3);
My goal was to turn off peripherals powered by the "3.3V".  But you said above that that is not a good thing to do since if the OLED (running off the "3.3V" header pin) is not powered then the I2C bus (shared with the AXP) will not work?

lewisxhe, 2024-08-15:
For V1.1, DC1 is the power supply for OLED, and DC3 is the power supply for the core of ESP32. However, since both of them share the I2C bus, if DC1 is turned off, the I2C bus will become invalid and cannot be accessed any more, so it cannot be turned off.

In V1.2, there is only one DC1 to power the ESP32 core. The pins of the OLED screen and DC1 are in the same place. They are together. Their specific functions can be seen in the provided code.
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/8228c930fa56d6046685b138f7f4cca4c7b93818/examples/RadioLibExamples/Transmit_Interrupt/LoRaBoards.cpp#L120

william67, 2024-09-09:
Hi I am also trying to figure out how this works exactly. Basically I want the oled display (1.3 shield) still to work in battery mode, but it is now turned off (black screen) while the esp32 itself remains working (still sending signals). How can I use the screen in Battery mode?

william67, 2024-09-09:
Are you sure that the schematic LilyGo_TBeam_V1.2.pdf is the right one, vcc_2.5 going to the oled shield it still connected to LX3 there?!? The name in the schema T22_GPS_V1.1»»³ÉAXP2101 must be v1.2 I supose.

lewisxhe, 2024-09-10:
The LX3 inductor has been removed from the schematic diagram. It can be seen from the actual hardware that there is only one inductor in the entire power supply, which is DC1. The NC inductor of DC3
[image]

moshe-braner, 2024-09-10:
So both the ESP32 and the "3.3V" solder tab are fed by DC1?

The new schematic is not posted yet?  The one posted here:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/8228c930fa56d6046685b138f7f4cca4c7b93818/schematic/LilyGo_TBeam_V1.2.pdf
still shows the LX3 inductor.

lewisxhe, 2024-09-11:
Yes! The schematic is the same, but the LX3 inductor has been removed during production.

moshe-braner, 2024-09-11:
OK.  The "3.3V" solder tabs, pins 7 and 11 in U14, are said in the schematic to connect to "VCC_2_5V", which is DCDC3, so something else must have changed too?  Maybe in the current boards R54 is "OR" rather than "NC"?  Otherwise there would be no power at those pins (e.g., for the OLED).

closed 2024-10-26 by github-actions[bot]