Xinyuan-LilyGO/LilyGo-LoRa-Series #191: the new T3 S3 E-paper board

moshe-braner, 2024-07-25:
The sales page for this new board does not let one choose the radio type or band, so either there is an error in the page, or this is not ready for sale yet?

What is item #5 in the photo of the included components - is it an external wifi antenna?  Does the board not have an on-board wifi antenna?

moshe-braner, 2024-07-25:
I wish you would offer a variant with the ESP32-S3-MINI-1U-N8 MCU, since for my purpose PSRAM is not really necessary, but there is not quite enough room in the flash (if using OTA partitioning).

lewisxhe, 2024-08-03:
Your question has been reported

closed 2024-09-18 by github-actions[bot]

beachmiles, 2025-03-26:
It looks like it's the 2.4ghz wifi/Bluetooth antenna as I don't see one on the board and I was getting very bad Bluetooth range before I connected the antenna. Wish they put a small onboard antenna on with the option to use the external antenna.